carla-simulator/carla #3051: Merging multiple point cloud from depth camera

Doodle1106, 2020-07-13:
Hi,

I am running the master branch of Carla and I'm trying to build a lane map from depth camera and semantic segmentation camera.
The process I followed is:

1. extract lane lines from segmentation;
2. project segmentation result to camera frame using pinhole camera model, call it p3d_camera;
3. transforming p3d_camera to the world frame by calling "camera.get_transform()".
4. combine multiple captures and build the lane map;

however, I think there is something wrong with the coordinate frame because i got weird result after merging 4 captures:

[image]
[image]

Is there anything I am missing here?
Thanks for trying to help!
Joey

Doodle1106, 2020-07-13:
the camera to world transform is from : Twv = self.camera.get_transform()

is it the right function to use?

germanros1987, 2020-07-14:
@DSantosO could you help with this, please?

DSantosO, 2020-07-14:
Sure, I will look into it.

Doodle1106, 2020-07-22:
I can send a copy of my code if that helps. 
just let me know. Thanks for trying to help.

DSantosO, 2020-07-22:
@Doodle1106 Yes, sure. Post it here or send it to me to daniel.santos@osvf.org and I will look into it as soon as I can.

closed 2020-09-21 by Doodle1106

aczy118, 2021-07-15:
Hi there,

How did you project extracted lane lines information from segmentation camera to depth camera? Can you please share a snippet of your code?

aczy118, 2021-07-15:
> I can send a copy of my code if that helps.
> just let me know. Thanks for trying to help.

Hi there,

How did you project extracted lane lines information from segmentation camera to depth camera? Can you please share a snippet of your code?