carla-simulator/carla #3053: Masking road and roadlines using semantic segmentation

RadioGarret, 2020-07-14:
hi there!

I would like to ask you how to mask road and road lines using semantic segmentation. 

I'm doing multi-threading and here're some codes

def MultiThreading(queue):
     while exitFlag is not True:
          queue_item = queue.get()
          queue_item = queue_item[: ,: ,2] # this converts the image to grascale
          ...

Does anyone know how to mask road and road lines only? 
I've checked out the below images already

[image]

[image]

Doodle1106, 2020-07-14:
I'm doing someting similar, and here is my implementation to extract lanes:

``` 
    # see https://github.com/carla-simulator/carla/blob/master/LibCarla/source/carla/image/CityScapesPalette.h
    def extract_region(self, segmentation):
        
        H, W, C = segmentation.shape
        canvas = np.zeros((H, W), dtype=np.uint8)
        for v in range(H):
            for u in range(W):
                if segmentation[v, u, 0] == 50:
                    canvas[v, u] = 255
        
        return canvas
```

germanros1987, 2020-07-14:
I don't believe this is the right place for this type of question. The forum is intended for that: [forum.carla.org](forum.carla.org). Please, create a question there. Leave GitHub for questions related to API behaviors and doubts. Use the forum for more general questions

closed 2020-07-14 by germanros1987