carla-simulator/carla #3056: 'make import' only imports the open drive file but no models

ZhengZhongyuan11, 2020-07-14:
Hi guys, I have some problem with the command 'make import'. When I try to import my own map, which is created from RoadRunner,  by using make import, it only imports the open drive file, but no model was imported in the UE4 editor. In another word, it will generate a series of waypoints with an empty world. 

Dose anyone know what happens here? 

Thanks!

ZhengZhongyuan11, 2020-07-14:
[2020.07.14-18.56.17:527][  0]LogCore: Error: Invalid input P=360.648926 Y=-nan R=0.000000 to FRotator::Quaternion - generated NaN output: X=nan Y=nan Z=nan W=-nan

I got this log message, but how could know where this error happens?

germanros1987, 2020-07-20:
@marcgpuig since you are working on the import pipeline these days, could you take a look at this, please?
@ZhengZhongyuan11 could you prepare your map to be shared with @marcgpuig ?

ZhengZhongyuan11, 2020-07-27:
@germanros1987 Yeah, it is available, but could you leave me your email address, so that I could send it to you?

marcgpuig, 2020-08-19:
Hi, @ZhengZhongyuan11 
Can you send it to marc.garcia@osvf.org?

closed 2020-11-08 by stale[bot]