carla-simulator/carla #3059: Carla cpp client for Android OS

meitalcohenyaacobi, 2020-07-15:
Hi , 
 we would like to generate carla c++ client for Android OS, 
Do you support Android OS (client side)? 
Do you support both x86/x86_64 and Arm/Aarch64 (client side)? 
Thank you.

germanros1987, 2020-07-20:
Hi @meitalcohenyaacobi, 

CARLA does not support a build for Android. You are welcome to play with the build process to generate a client for Arm architectures. We would be happy to hear about your progress.

Best regards.

meitalcohenyaacobi, 2020-07-20:
Hi @germanros1987  
Thanks a lot.

I understand that I can't run CARLA simulator on android. but what I want to do is running a self-implemented client for connecting to the simulator server and receive sensory data from the simulator. The client run on remote machine and connect to CARLA via RPC calls. As far as I understand from the examples, the only dependency of such client is rpcLib. I found that I can compile rpcLib to Android, is there any other dependency that prevent me from develop such client for Android?

germanros1987, 2020-07-30:
Not that I am aware of. @bernatx do you see any potential blockers?

closed 2020-08-29 by germanros1987