carla-simulator/carla #3064: LibCarla C++ headers missing in release

johschmitz, 2020-07-15:
It seems there is no lib and include folders in the tar ball release which contain the LibCarla and are required to build the Client.
Could you include them in the next release?

germanros1987, 2020-07-20:
@johschmitz the artifacts of the release are pre-generated binary builds. You can get all the libs and includes by cloning the right tag. This is intentional.

closed 2020-07-20 by germanros1987

johschmitz, 2020-07-20:
What about the Debian/Ubuntu package?Normally any other software is structured in a way that it is possible to use also precompiled libraries and build against them. Wouldn't it be nice if one could do a
sudo apt install carla-simulator-dev
and then be able to compile custom cpp client with the Ubuntu 18.04 default GCC compiler? 
That was my idea behind opening this issue.
It would make everything much easier for the consumers of the CARLA client libs. 
This is of course related to #3065.