carla-simulator/carla #3065: CppClient example Clang vs GCC and build system

johschmitz, 2020-07-15:
I would like to use GCC 7 to build the CppClient example.
GCC 7 is also specified in the included Makefile. However it seems it wants to use Clang to build LibCarla.
I doubt that we can even guarantee ABI compatibility between GCC and Clang.
What is the correct way to build the example with GCC?

I am a bit confused since the CARLA build system seems to follow be a big monolithic mashup approach instead of a classical modular CMake approach. Would be cool if it would be possible to build the library standalone and then later link against it to build a client. After all to really build my own client I need to build it independently.
What is the recommended way to build a custom CppClient?

wx9698, 2020-07-17:
Feel free to take a look at this [repo](https://github.com/wx9698/carlaviz/tree/master/backend). I use cmake to manage my building process and use both GCC and Clang to build the entire project. It might be helpful to you. 
The LibCarla is under the "third_party" folder.

wx9698, 2020-07-17:
Yes, you can build the entire client example with gcc only.

germanros1987, 2020-07-20:
@bernatx could you also provide some clarity here? Have you been able to build 0.9.9 with gcc-7? Also, we should probably make our cmake more modular, as suggested...

johschmitz, 2020-07-20:
@germanros1987 @bernatx have you considered using Conan? It would be very awesome to be able to resolve all dependencies through Conan Center/Bintray and then build a CARLA client using a Conan recipe.
If the CMake is structured in a good way it should work with and without Conan. A little wrapper CMakeLists.txt at the top layer can achieve this.

closed 2020-09-29 by stale[bot]