carla-simulator/carla #3070: Semantic Tag for new props does not work Carla 0.9.9

Jabkowsky, 2020-07-16:
Window version build from source ( 0.9.9.3 ) 
I have add the new props to .json file : ( Default.Package.json )
Example:
		{
			"name": "PedCross",
			"path": "/Game/Carla/Static/Props/Street/SM_PedCross.SM_PedCross",
			"size": "Small",
			"tag": "TrafficSigns"
		},

After build I can spawn "PedCross" sign as i need, but this "actor" does not appear on camera semantic segmentation as "TrafficSigns". ( tag ). All new props added are visible on segmentation as Unlabeled = 0 even if i specified the tag in .json file.

Can you help me to fix this issue ? Am I doing something wrong ?

germanros1987, 2020-07-20:
@marcgpuig since you are working on this, could you please provide a solution?

Jabkowsky, 2020-07-20:
EDIT: i tried with properly tagged stop sign in town10hd and add the same stop sign to jason file with same tag. It look like this
[image]
Originally added stop sign to town is segmented with TAG = 12 correctly but spawned same sign :
		{
			"name": "Stop",
			"path": "/Game/Carla/Static/TrafficSigns/Stop/SM_stopSign.SM_stopSign",
			"size": "Small",
			"tag": "TrafficSigns"
		}
Is not. It appears only with TAG = 0 :
[image]

And now difference in carlalic ( c++ ) i found :
Original stop sign is such actor -> 
[image]

It has and correct tag "semantic_tags"[1] = 12 

But props actor added to map while "playing" ->
[image]

has empty semantic_tags property.

closed 2020-09-29 by stale[bot]