carla-simulator/carla #3072: [Request] Video showing how to do Windows Build

wbadry, 2020-07-16:
Hello,
I read the document showing how to do [a Windows Build](https://carla.readthedocs.io/en/latest/build_windows/). However, for the Unreal Engine **UE4_ROOT** variable, it seemed to be written in a hurry as the statement of "the desired UE4 installation" is unclear at all.

I am aware that most of CARLA users are on Linux machines. However, I would appreciate it if someone could have time to post a video showing the installation process on Windows. It would be much helpful to everyone in the future.

Also, what is the difference between the Windows release of CARLA and cloning the repository? Thanks

germanros1987, 2020-07-20:
@sergi-e could you please take care of making this part more clear in the documentation?

@wbadry in the release we create a binary build, while the repository is just the raw project.

sergi-e, 2020-08-31:
Hi, @wbadry !

We will update the build docs as soon as possible, hopefully with detailed videos on how to make the build for both Linux and Windows :) 

Regarding the difference between the Windows release and the repository... it is essentially what german1987 said. Using the release you have all the features that the CARLA team developed by the time of the release. By cloning the repository, you get the current state of CARLA that can be updated anytime.

wbadry, 2020-09-02:
Thanks @sergi-e and @juaxix for your response. I highly appreciate it

closed 2020-11-14 by stale[bot]