carla-simulator/carla #3073: Vehicle does not move when I set carla.WheelPhysicsControl

mehmor, 2020-07-16:
I  added friction to the vehicle's wheels using the following code:

        front_left_wheel  = carla.WheelPhysicsControl(tire_friction=0.01, damping_rate=1.0, max_steer_angle=70.0, radius=30.0)
        front_right_wheel = carla.WheelPhysicsControl(tire_friction=0.01, damping_rate=1.5, max_steer_angle=70.0, radius=25.0)
        rear_left_wheel   = carla.WheelPhysicsControl(tire_friction=0.01, damping_rate=0.2, max_steer_angle=0.0,  radius=15.0)
        rear_right_wheel  = carla.WheelPhysicsControl(tire_friction=0.01, damping_rate=1.3, max_steer_angle=0.0,  radius=20.0)

        wheels = [front_left_wheel, front_right_wheel, rear_left_wheel, rear_right_wheel]
        # Change Vehicle Physics Control parameters of the vehicle
        physics_control = carla.VehiclePhysicsControl()
        
        physics_control.torque_curve = [carla.Vector2D(x=0, y=400), carla.Vector2D(x=1300, y=600)]
        physics_control.max_rpm = 10000
        physics_control.moi = 1.0
        physics_control.damping_rate_full_throttle = 0.0
        physics_control.use_gear_autobox = True
        physics_control.gear_switch_time = 0.5
        physics_control.clutch_strength = 10
        physics_control.mass = 10000
        physics_control.drag_coefficient = 0.25
        physics_control.steering_curve = [carla.Vector2D(x=0, y=1), carla.Vector2D(x=10, y=0.5)]
        #physics_control.steering_curve = [carla.Vector2D(x=0, y=1), carla.Vector2D(x=100, y=1), carla.Vector2D(x=300, y=1)]
        physics_control.wheels = wheels
        # Apply Vehicle Physics Control for the vehicle
        vehicle1.apply_physics_control(physics_control)

But when I set the velocity of vehicle1, it does not move in the street. I just jumping up and down.

I appreciate if some one help me with it.

germanros1987, 2020-07-20:
@DSantosO since you are the new master of physics, could you take a look into this? I am also adding @marcgpuig in case he has some insights. @manishthani if you are alive and want to comment, feel free...

closed 2020-09-29 by stale[bot]

lijecaru, 2021-03-18:
Is there any update on this? I have the same problem