carla-simulator/carla #3076: [0.9.7] Cant find "dependencies\lib" and also the .egg files and thus unable to run Python scripts in carla like "spawn_npc.py"

TheMaveric, 2020-07-17:
```
D:\carla-0.9.7\LibCarla\source\carla\client\detail\Client.cpp(10): fatal error C1083: Cannot open include file: 'carla/Version.h': No such file or directory [D:\carla-0.9.7\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client.vcxproj]
  Client.cpp
D:\carla-0.9.7\Build\boost-1.69.0-install\include\boost/asio/buffer.hpp(33): warning C4005: 'BOOST_ASIO_ENABLE_BUFFER_DEBUGGING': macro redefinition [D:\carla-0.9.7\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
  D:\carla-0.9.7\LibCarla\source\carla\client\detail\Client.cpp: note: see previous definition of 'BOOST_ASIO_ENABLE_BUFFER_DEBUGGING'
D:\carla-0.9.7\LibCarla\source\carla\client\detail\Client.cpp(10): fatal error C1083: Cannot open include file: 'carla/Version.h': No such file or directory [D:\carla-0.9.7\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]
-[BuildLibCarla]: LibCarla server has been successfully installed in "D:/carla-0.9.7/Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies"!
-[BuildLibCarla]: LibCarla client has been successfully installed in "D:/carla-0.9.7/PythonAPI\carla\dependencies"!
-[BuildPythonAPI]: [Batch params]: --rebuild
Building Python API for Python 3.
Traceback (most recent call last):
  File "setup.py", line 131, in <module>
    ext_modules=get_libcarla_extensions(),
  File "setup.py", line 91, in get_libcarla_extensions
    libs = [x for x in os.listdir('dependencies/lib') if any(d in x for d in required_libs)]
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'dependencies/lib'

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:/carla-0.9.7/PythonAPI\carladist"!
```

But the list folder is not created and also no egg file is found.

germanros1987, 2020-07-20:
@TheMaveric are you finding these issues in the 0.9.9 version too? Could you please check?

TheMaveric, 2020-07-20:
I do not get these in the prebuilt 0.9.8 version but I need to use UE4 and 0.9.7.. I think the .egg file is the issue.. Is there a link to the .egg file for 0.9.7 windows?

closed 2020-09-29 by stale[bot]