carla-simulator/carla #3079: labels for LiDAR data

tiger-bug, 2020-07-17:
Good afternoon,

I was curious if there will be a label feature added to the LiDAR sensor.  I know several other people have posted about this in [issue 1751](https://github.com/carla-simulator/carla/issues/1751) and [issue 516](https://github.com/carla-simulator/carla/issues/516), and that individual has a version of carla with labels for LiDAR.  Is this something that the developers are looking to do in the future?  I am using Carla, carla-ros-bridge, and SLAM to create point clouds for DL training.  Thank you!

germanros1987, 2020-07-20:
Future versions of CARLA will provide ground truth for point clouds. We will visit this topic during next releases. If this has a high priority for you and you have some bandwidth, perhaps you could contribute a solution...? Contributions are very appreciated!

germanros1987, 2020-07-30:
@tiger-bug good news. This feature has been recently developed and will be part of 0.9.10, expected in a few days.

tiger-bug, 2020-07-30:
Thank you!!! That is awesome!

marcgpuig, 2020-07-31:
Closing the issue since this has entered in master in #3131!

closed 2020-07-31 by marcgpuig

visakhchikku, 2023-10-05:
Is it possible to save lidar data as .pcap file using carla?