carla-simulator/carla #3080: v0.9.9 on Ubuntu 18.04. Segmentation fault. Malloc error. Signal 11.

RansML, 2020-07-18:
I have been running carla_release_0.9.5 on Ubuntu 18.04 (with cuda 10.0) without having any trouble. I just downloaded the latest Carla binary (carla_release_0.9.9).

Now `DISPLAY= ./CarlaUE4.sh ` gives the following error. 

> 4.24.3-0+++UE4+Release-4.24 518 0
> Disabling core dumps.
> Signal 11 caught.
> Malloc Size=65538 LargeMemoryPoolOffset=65554 
> CommonUnixCrashHandler: Signal=11
> Malloc Size=65535 LargeMemoryPoolOffset=131119 
> Malloc Size=112464 LargeMemoryPoolOffset=243600 
> Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
> Segmentation fault (core dumped)

Running `DISPLAY= ./CarlaUE4.sh -opengl3 ` shows nothing after printing the first two lines:
> 4.24.3-0+++UE4+Release-4.24 518 0
> Disabling core dumps.
It looks like the GPU is trying to run something.

germanros1987, 2020-07-20:
@RansML could it be that your driver doesn't support vulkan or that you haven't installed the vulkan dependencies? It seems that it is working perfectly using opengl (yes, that is what those 2 lines mean...everything is working fine). Could you check your driver, dependencies, etc?

RansML, 2020-07-20:
Just to summarize,
- Carla v.0.9.5 (Ubuntu binaries) works when I run `DISPLAY= ./CarlaUE4.sh`. 
- Carla **v.0.9.9** (Ubuntu binaries) does not work when I run `DISPLAY= ./CarlaUE4.sh` or `DISPLAY= ./CarlaUE4.sh -opengl`.
I have a NVIDIA GTX1060 GPU on a Razer Blade. 

I also tried on a different laptop on Ubuntu 20.04. I get exactly the same error.

Axel1092, 2020-07-20:
Hi @RansML,
Using the `DISPLAY= ...` is intended to disable rendering (no window will appear) and it only works with the `-opengl` option. After running `DISPLAY= ./CarlaUE4.sh -opengl` have you tried to connect to the server? Try running the `test_connection.py` script in the PythonAPI/util folder.

RansML, 2020-07-20:
@Axel1092 When I remove `DISPLAY=` it loads the CarlaUE4 window and I can interact with it using my mouse. However, nothing shows on the CLI after,

> 4.24.3-0+++UE4+Release-4.24 518 0
> Disabling core dumps.

`test_connection.py` fails with the message 

> Failed to connect to 127.0.0.1:2000.

When I run v.0.9.**5**, lots of print messages are typically displayed on the CLI before I can actually connect successfully.

wendersonj, 2020-08-24:
Using `DISPLAY= ./CarlaUE4.sh -quality-level=Low -opengl` worked for me.
GPU: GeForce MX130 (2GB)

Cheers

closed 2020-11-08 by stale[bot]

nivolog, 2022-10-21:
Currently facing the same problem on ASUS FX705DT (Ryzen 3550H, GTX 1650 TI, 16 GB RAM). Ubuntu 20.04, using CARLA package (/opt/carla-simulator) and I almost always have exactly the same error, while trying to run CARLA. However, sometimes i have a success and CARLA launches normally. More specifically, I use BARK CARLA interface, it's open project, you can find it on github. But they use the usual Python API and launch simulator also from opt folder. I barely can see the difference in situations between my successfull launches and errors, my idea is that I have lack of GPU memory. Hope that I'm wrong and there will be some solution to that!