carla-simulator/carla #3081: Is that something wrong about IMU document and code in manual_control.py about compass?

JarvanI, 2020-07-20:
Hi, everyone, I found somthing that made me confused and I hope someone can help me.
Here is a piece of code in manual_control.py, from line 528 to line 532
`        compass = world.imu_sensor.compass `
       ` heading = 'N' if compass > 270.5 or compass < 89.5 else ''`
     `   heading += 'S' if 90.5 < compass < 269.5 else ''`
    `    heading += 'E' if 0.5 < compass < 179.5 else ''`
       ` heading += 'W' if 180.5 < compass < 359.5 else '' `
I got two question about the code.
First , from the code I found that the unit of compass is degrees, not radians. But in Carla Simulator Document ,the unit of compass is radians. Is the document wrong?
[image]

Second , is the east and west direction opposite to normal？If the right hand direction is north, then up is west, and down is east.  So in my opinion,     `    heading += 'W' if 0.5 < compass < 179.5 else ''`and ` heading += 'E' if 180.5 < compass < 359.5 else '' ` is right.

germanros1987, 2020-07-24:
@marcgpuig could you take a look at this issue?

marcgpuig, 2020-08-18:
Hi @JarvanI 
> If the right hand direction is north, then up is west, and down is east

I'm not sure to correctly understand your point.
If I've done it right I'm following the angle of a standard compass, where North is at 0° (or 360°), East at 90°, South at 180° and Weast at 270°.

closed 2020-11-08 by stale[bot]