carla-simulator/carla #3083: module 'carla' has no attribute 'Client'

uniwuppes, 2020-07-20:
Hello,

I'm having trouble with the Python 3 module of carla.
I compiled it with `make PythonAPI` and it runs through without any errors.

I can also import the module successfully. 
But the imported object does not have any child classes .

So when I'm trying to run an example script, I get this message
`module 'carla' has no attribute 'Client'`

The Python 2 version works just fine.

I don't have any other carla module installed via pip.

aqvia, 2020-07-21:
I have the same problem.
Carla 0.9.9 / Ubuntu 18.04 / Python 3.6.9

```
$ python3 spawn_npc.py 

done.
Traceback (most recent call last):
  File "spawn_npc.py", line 274, in <module>
    main()
  File "spawn_npc.py", line 92, in main
    client = carla.Client(args.host, args.port)
AttributeError: module 'carla' has no attribute 'Client'
```

Trying with Carla master, I got the following output.

```
$ python3 spawn_npc.py 
Traceback (most recent call last):
  File "spawn_npc.py", line 26, in <module>
    from carla import VehicleLightState as vls
ImportError: cannot import name 'VehicleLightState'
```

yasser-h-khalil, 2020-07-23:
Same issue here

germanros1987, 2020-07-24:
@bernatx could you take a look at this issue?

Axel1092, 2020-07-29:
Hi all, we have found that there is an issue with the latest version of the setuptools library in python3 that builds the *.egg files with the PythonAPI. Using an earlier version of the library solver the issue, this command installs the working version `pip3 install -Iv setuptools==47.3.1`.

aqvia, 2020-07-30:
@Axel1092 It works fine. Thank you for the information.

uniwuppes, 2020-08-14:
@Axel1092 That fixed it for me too. Thanks.

closed 2020-08-14 by uniwuppes

Malathi15, 2020-08-19:
Hi @Axel1092 
I got an while tried to import sensor from carla. 
I have got this error
```
 python3 module_7.py 
In /home/ioz/.local/lib/python3.6/site-packages/matplotlib/mpl-data/stylelib/_classic_test.mplstyle: 
The text.latex.preview rcparam was deprecated in Matplotlib 3.3 and will be removed two minor releases later.
In /home/ioz/.local/lib/python3.6/site-packages/matplotlib/mpl-data/stylelib/_classic_test.mplstyle: 
The mathtext.fallback_to_cm rcparam was deprecated in Matplotlib 3.3 and will be removed two minor releases later.
In /home/ioz/.local/lib/python3.6/site-packages/matplotlib/mpl-data/stylelib/_classic_test.mplstyle: Support for setting the 'mathtext.fallback_to_cm' rcParam is deprecated since 3.3 and will be removed two minor releases later; use 'mathtext.fallback : 'cm' instead.
In /home/ioz/.local/lib/python3.6/site-packages/matplotlib/mpl-data/stylelib/_classic_test.mplstyle: 
The validate_bool_maybe_none function was deprecated in Matplotlib 3.3 and will be removed two minor releases later.
In /home/ioz/.local/lib/python3.6/site-packages/matplotlib/mpl-data/stylelib/_classic_test.mplstyle: 
The savefig.jpeg_quality rcparam was deprecated in Matplotlib 3.3 and will be removed two minor releases later.
In /home/ioz/.local/lib/python3.6/site-packages/matplotlib/mpl-data/stylelib/_classic_test.mplstyle: 
The keymap.all_axes rcparam was deprecated in Matplotlib 3.3 and will be removed two minor releases later.
In /home/ioz/.local/lib/python3.6/site-packages/matplotlib/mpl-data/stylelib/_classic_test.mplstyle: 
The animation.avconv_path rcparam was deprecated in Matplotlib 3.3 and will be removed two minor releases later.
In /home/ioz/.local/lib/python3.6/site-packages/matplotlib/mpl-data/stylelib/_classic_test.mplstyle: 
The animation.avconv_args rcparam was deprecated in Matplotlib 3.3 and will be removed two minor releases later.
/home/ioz/Desktop/CarlaSimulator/PythonClient/live_plotter.py:8: MatplotlibDeprecationWarning: 
The matplotlib.backends.tkagg module was deprecated in Matplotlib 3.0 and will be removed in 3.2.
  import matplotlib.backends.tkagg as tkagg
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "module_7.py", line 33, in <module>
    from carla            import sensor
ImportError: cannot import name 'sensor'
```
I have installed setuptools as you said `pip3 install -Iv setuptools==47.3.1` but still got the same error.

Please help me to solve this.
Thanks in Advance
Malathi

Axel1092, 2020-08-26:
Hi @Malathi15, we'll take a look to this problem although it seems unrelated to the previous problem.

guruvishnuvardan, 2020-08-31:
@Axel1092 - Thanks for the Support. We will try as advised and get back to You.

man-csl, 2020-09-02:
I also get the same error, but installed carla using the deb package. How can I fix it?

Axel1092, 2020-09-04:
Hi @man-csl, maybe you are not using the correct python version. We support python 3.7 for our packaged builds, try installing it (`sudo apt-get install python3.7`) and see if it works.

KuldeepDileep, 2021-05-05:
> Hi @Malathi15, we'll take a look to this problem although did you rebuild carla after running `pip3 install -Iv setuptools==47.3.1`? Try with `make clean` and `make PythonAPI` and see if something else fails in the process.

@Axel1092 it didn't work for me
**Commands:**
pip3 install -Iv setuptools==47.3.1
sudo python3 module_7.py
**Error**
ImportError: cannot import name 'sensor' from 'carla' (/home/kuldeep/Downloads/CARLA_0.9.10/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64/carla/__init__.py)

itserror404, 2023-07-24:
hii, how did you fix this?