carla-simulator/carla #3084: New maps with road construction zone: How to generate reference routes manually?

bosu3076, 2020-07-20:
Hello guys. I'm trying to make new maps with some road constructions, that means I need to generate the routes by myself, since the automatically generated routes by OpenDRIVE Actor in UE4 could not be used. However, I tried RoutePlanner tool, I am quite confused about how does it work. 
[image]
In this example, I drew a number of routes with RoutePlanner- the white splines, and kept some default routes- red ones, which were generated automatically by OpenDRIVE Actor.
My problems are:
1. Why the default and self-defined route look different? Is there anything I set wrong?
2. Those routes could not be followed by the vehicles with autopilot. The documentation about Traffic Manager has no description of this part. I wonder  whether the Traffic Manager could be used in this kind of situation.

Version Infos:
-Ubuntu 16.04
-Carla 0.9.8
-Unreal Engine 4.22
-RoadRunner 2020a

Thank you!

germanros1987, 2020-07-24:
@marcgpuig do you have some cycles to help with a simple explanation of the RoutePlanner and/or the OpenDRIVE Actor?

XieJinLiang, 2020-08-04:
What software are you using? roadrunner? Why does the interface look different from my roadrunner?

bosu3076, 2020-08-04:
> What software are you using? roadrunner? Why does the interface look different from my roadrunner?

Hi, actually this is a screenshot of Unreal Engine 4 https://www.unrealengine.com/en-US/. I have imported the fbx-files into UE4 and want to edit them with  Carla plugins. However, it doesn't work well. These days I changed my plan generating routes directly in Roadrunner.

marcgpuig, 2020-08-18:
Hi @bosu3076, sadly this is not possible within Unreal Engine, we strongly rely on OpenDRIVE right now to retrieve information from the road. The `RoutePlanners` and the `OpenDRIVE Actor` are not used anymore but for automatically placing the spawn points and visualizing the shape of the used OpenDRIVE. You need to do everything with RoadRunner.

closed 2020-11-08 by stale[bot]

ehsankf, 2023-12-13:
How could we add work zone construction to the existing maps?