carla-simulator/carla #3085: Carla wastes CPU ressources in sync mode between ticks

hh0rva1h, 2020-07-20:
Consider the following code snippet:
```
import carla
import time

client = carla.Client("localhost", 2000)
world = client.load_world("Town04")

settings = world.get_settings()
settings.fixed_delta_seconds = 0.05
settings.synchronous_mode = True
world.apply_settings(settings)

world.tick()
time.sleep(120)
world.tick()
```
Between the ticks Carla is frozen so I would expect it not to use much CPU resources, however actually Carla is always hogging the CPU:

[image]

It would be great if Carla wouldn't waste resources when there is no simulation to do.

germanros1987, 2020-07-24:
Hi,

@hh0rva1h we will definitely try to keep improving the performance of CARLA and reducing the amount of computation needed. Now, while you are not ticking the server, the server is still doing stuff. At a minimum is listening for other possible clients (there could be more than 1 client trying to get things done). 

We will try to make this aspect more efficient in the future. I will tag this issue as a feature request.

hh0rva1h, 2020-09-28:
@germanros1987 Could you do something about stale bot closing issues that are triaged other than question? Imho it does not make much sense to close a issue triaged as bug or feature request automatically.

closed 2020-12-25 by stale[bot]