carla-simulator/carla #3086: Can not genereate pedestrian navigation info with RecastBuilder

little9999, 2020-07-21:
Hi,

I'm using carla 0.9.9.2 under ubuntu 18.04.

Today I created a custom road map with RoadRunner, and then I follow the doc https://carla.readthedocs.io/en/latest/tuto_D_generate_pedestrian_navigation/ to generate the pedestrian navigation info.

The RecastBuilder log sounds good :
Loading...
Building...
Saving...

However I can not find the generated *.bin file.

**So I double check with the offical maps: Town01, Town02 and Town04.
I export the OBJ file and then launch RecastBuilder, everytime I have the same log "loading building saving", but there is only the Town04.bin is generated.**

Bests,
Tao

germanros1987, 2020-07-24:
@bernatx could you have a look at this issue?

bernatx, 2020-07-24:
Hi, probably the map is too big to be processed. You can follow the following procedure. Just select all the meshes that involves in pedestrian navigation and deselect all that are not related, like the landscapes where the pedestrians can not navigate, the sky box...then export with the CARLA exporter, and try again with the new .obj.

charlie-wartnaby, 2020-08-17:
I'm afraid I opened a new issue when I should have added to this one, see #3189 .

I'm not even interested in pedestrian navigation, but did want the traffic lights set up so that I could address them in simulation. Even with an extremely simple RoadRunner model (one crossroads) I had the same problem. I did a little debugging, see #3189, but have not yet solved it. I do not think it is anything to do with the size of the model.

It would be extremely helpful having an open-source example map that we could practise the toolchain on (at least starting with an .xodr + .fbx (+.xml)? set from RoadRunner, I understand the raw asset files are proprietary to them).

I don't think I've got permissions to close #3189 now I've created it, but please feel free to do so, and I can copy my notes from there to this one assuming that's better.

Many thanks,

Charlie Wartnaby, Applus IDIADA, UK

charlie-wartnaby, 2020-08-17:
[Here is what I put in #3189 on the same topic.]

**RecastBuilder emits no .bin file at 'Generate pedestrian navigation' step of map import**

Dear all, I hope this is an appropriate question. I have not found the answer elsewhere. I have been struggling for days to get my own model with working traffic lights into CARLA, starting from RoadRunner.

Really I would like to start with the source RoadRunner files used for the 'official' assets to see if they work for me. Are they available anywhere, or are they private? Or at least the exported .xodr/.fbx/xml files, even if the original .rrscene and assets must be private because of RoadRunner licensing on the assets. Then I could figure out if I have a toolchain issue, or the problem is with my own model. ANY example model someone else has succeeded with would be great.

I have built CARLA from source starting from a fairly recent commit 8557c0b (15Jul2020) and have tried both Ubuntu 18.04 and Windows 10.

I have built a model in RoadRunner with a simple auto-signalised crossroads and one crosswalk. I have followed the instructions at https://carla.readthedocs.io/en/latest/tuto_A_add_map/#map-ingestion-in-a-build-from-source (most recently, have also tried RoadRunner plugin, the deprecated import method, docker image failed to build for me near end, carla-map-editor build also failed but I guess that's out of date now?). The closest I got was some animated traffic lights using the RoadRunner plugin in Ubuntu (not Windows), but that doesn't give me traffic light objects accessible via the PythonAPI, which I want to use with scenario_runner and OpenSCENARIO scripts.

I have also found the super-helpful video https://www.youtube.com/watch?v=mHiUUZ4xC9o presented by Bernat Pina (@bernatx ?).

Both on Ubuntu and Windows, RecastBuilder does not build any .bin file for me. I have debugged a bit in Visual Studio 2017 and it fails because:

    dtCreateNavMeshData() returns false because:
    if (!params->vertCount || !params->verts) is true
    vertCount is zero, though verts is non-NULL
    Likewise params->polyCount is 0 though polys is non-NULL
    So sample->handleBuild(); leaves us with m_navMesh=NULL
    then sample->SaveAll() quietly fails

I tried hacking some likely non-zero numbers into those counts in the debugger to get a .bin file, but that didn't get me any traffic lights in the model.

I feel it would be very valuable if there was at least one very simple standard example map asset where all the source was available, so newbies like me have something to experiment with to figure out what is going wrong! (E.g. the model used by Bernat in that video.)

Thank you in advance for any tips!

Charlie Wartnaby, Applus IDIADA, UK

closed 2020-11-08 by stale[bot]

ZJH166018, 2021-07-02:
> Hi, probably the map is too big to be processed. You can follow the following procedure. Just select all the meshes that involves in pedestrian navigation and deselect all that are not related, like [...]

Hi, I have tried this method, but it does not work.