carla-simulator/carla #3090: Add a thermal camera

boiscljo, 2020-07-21:
Hi, 

an increasing number of autonomous cars are using thermal cameras, a great new feature would be to include them as a new sensor in Carla. Even if it's just applying thermal-like textures to object on a different rgb camera.

germanros1987, 2020-07-24:
Hi @boiscljo,

I think you are right. We would happily take your contribution. How long do you think it would take you?

boiscljo, 2020-07-27:
Hi @germanros1987, 
I'm trying to evaluate the time required but I can't manage to open a recent enough version of Unreal Engine to open Carla simulator project:"We can't open project since it's made with a more recent version of Unreal Engine". Once opened, and if my assumptions that UnrealEngine works the same as Unity, it should be a matter of two weeks. Adding a second texture to each object on a separate "layer". The new camera would only see the new textures on the same object and the old camera won't see the new textures. If Unreal allows textures to be per layer meaning a camera would only see textures from that layer.

patrick-han, 2020-10-26:
Hi @marcgpuig this sounds like a really interesting idea that I might like to explore further. 

So essentially the idea would be, based on lighting conditions and the label from a segmentation tag, we can develop shaders for each object in a scene to "color" them in the view of the thermal camera? 

For example, a car might be shaded as "very hot" in the day time given that it's made of metal. Maybe we can extract the material from the object and use that information too? Perhaps we could also use light temperature?

closed 2021-07-21 by stale[bot]