carla-simulator/carla #3091: Carla UE4 always crash

carlaUE4new, 2020-07-21:
I just installed Unreal engine (4.24.3) and the latest Carla from GitHub at this link (https://github.com/carla-simulator/carla). It seems Unreal engine is running fine and I can play some examples. Carla was installed based on instruction for Windows in this link (https://carla.readthedocs.io/en/latest/build_windows/). Almost at the end of installation (using make launch), it launched CarlaUE4, however, I am not able to play it. Whenever I clicked on Play button, it will crash Unreal Engine. 

Some basic setting for my computer
Operating system: Windows 10 X64 , 2 GPUs (Intel HD graphics 530 and NVIDIA Quadro M3000M, shared memory is 16GB, dedicated memory is 4GB, DirectX version is 12

Installation process
•	I have followed the installation instruction for Windows from above link
•	Get the assets from ContentVersion.txt I tried 3 different versions of assets including the latest, it seems it makes no difference

Compiling process
•	I started the X64 native tools command windows with admin and then run make launch
•	It took long time to build everything probably around 1 hour
•	It then launched CarlaUE4 project. Per the instruction from their website (“Press Play to start the spectator view and close the editor window to exit. “), I clicked the Play button, however, it always crashed from there.
•	I have tried all methods mentioned in the forum, but cannot solve this issue

Below are some error messages I got from the crash report
o	LogWindows: Failed to load 'aqProf.dll' (GetLastError=126)
o	LogWindows: File 'aqProf.dll' does not exist
o	LogWindows: Failed to load 'VtuneApi.dll' (GetLastError=126)
o	LogWindows: File 'VtuneApi.dll' does not exist
o	LogWindows: Failed to load 'VtuneApi32e.dll' (GetLastError=126)
o	LogWindows: File 'VtuneApi32e.dll' does not exist
o	LogConsoleResponse: Display: Failed to find resolution value strings in scalability ini. Falling back to default.

Lu7913, 2020-07-23:
Hi,
Happened to me recently, updating the assets helped, I saw some assets have been changed recently, try with latest or one before that.

germanros1987, 2020-07-24:
@sergi-e could you try to help this user, please?

carlaUE4new, 2020-07-24:
> @sergi-e could you try to help this user, please?

thank you, germanros1987. I got those above errors when I tried to install it and run it on my company's computer. I am not sure if there are some firewall issue or TCP port availability that could cause this problem. I tried to do the same thing tonight on my personal computer, it worked. So, could you please suggest what kind of firewall restriction or network policy could potentially cause those above errors?

Luque346, 2020-08-01:
Buy a PC

closed 2020-10-11 by stale[bot]