carla-simulator/carla #3092: 0.9.8 - Cannot create navigation bin file if import by Create one Blueprint asset

youngsend, 2020-07-22:
Hi, everyone.
I am new to carla. I am using 0.9.8 and I want to spawn walker on my own map.

What I have done:
When I import my test.fbx into carla Unreal Editor by _Create one Blueprint asset_, I cannot generate the pedestrian navigation bin file.
I try to import test.fbx into Editor by *Create Level Actors*, and I succeeded in generating bin file and spawning walkers on sidewalk.
However, I find that the latter import method loses RoadRunnerTrafficJunction information compared to the former method.

My question:
1. Can I generate pedestrian navigation bin file while importing by *Create one Blueprint asset*?
I think this should be possible, because all tutorials I found choose *Create one Blueprint asset*.
2. If I use *Create Level Actors*, will there be a big difference in the compiled package? 
I am worried about the lost of RoadRunnerTrafficJunction.

Thank you.

youngsend, 2020-07-22:
I deleted the Sidewalk StaticMesh in FbxScene and added it (it is in Static folder) again in World Outliner.
After that I succeeded in creating pedestrian navigation file and spawning walker on sidewalk.

However, I still want to ask why navigation bin file cannot be created with FbxScene holding the Sidewalk StaticMesh.

germanros1987, 2020-07-24:
@bernatx could you help with this?

bernatx, 2020-07-24:
Hi,
Have you any file to share so we can test it?
I suppose the FBX has something that has problems when it is imported to Unreal (we use the FBX importer from Unreal).

closed 2020-10-04 by stale[bot]