carla-simulator/carla #3093: Saving all camera sensor outputs to numpy array collectively

tgautam03, 2020-07-22:
Hi, I am using Carla 0.9.9.4 and running the script tutorial_ego.py 
```
import glob
import os
import sys
import time

try:
	sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
		sys.version_info.major,
		sys.version_info.minor,
		'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
	pass

import carla

import argparse
import logging
import random

def main():
	argparser = argparse.ArgumentParser(
		description=__doc__)
	argparser.add_argument(
		'--host',
		metavar='H',
		default='127.0.0.1',
		help='IP of the host server (default: 127.0.0.1)')
	argparser.add_argument(
		'-p', '--port',
		metavar='P',
		default=2000,
		type=int,
		help='TCP port to listen to (default: 2000)')
	args = argparser.parse_args()

	logging.basicConfig(format='%(levelname)s: %(message)s', level=logging.INFO)

	client = carla.Client(args.host, args.port)
	client.set_timeout(10.0)

	try:

		world = client.get_world()
		ego_vehicle = None
		ego_cam = None
		ego_col = None
		ego_lane = None
		ego_obs = None
		ego_gnss = None
		ego_imu = None

		# --------------
		# Start recording
		# --------------
		
		client.start_recorder('/home/tushar/work/carlaData/Town02/recorder/recording01.log')
		

		# --------------
		# Spawn ego vehicle
		# --------------
		
		ego_bp = world.get_blueprint_library().find('vehicle.tesla.model3')
		ego_bp.set_attribute('role_name','ego')
		print('\nEgo role_name is set')
		ego_color = random.choice(ego_bp.get_attribute('color').recommended_values)
		ego_bp.set_attribute('color',ego_color)
		print('\nEgo color is set')

		spawn_points = world.get_map().get_spawn_points()
		number_of_spawn_points = len(spawn_points)

		if 0 < number_of_spawn_points:
			random.shuffle(spawn_points)
			ego_transform = spawn_points[0]
			ego_vehicle = world.spawn_actor(ego_bp,ego_transform)
			print('\nEgo is spawned')
		else: 
			logging.warning('Could not found any spawn points')
		

		# --------------
		# Add a RGB camera sensor to ego vehicle. 
		# --------------
		
		rgb_cam_bp = None
		rgb_cam_bp = world.get_blueprint_library().find('sensor.camera.rgb')
		rgb_cam_bp.set_attribute("image_size_x",str(960))
		rgb_cam_bp.set_attribute("image_size_y",str(540))
		rgb_cam_bp.set_attribute("fov",str(105))
		rgb_cam_location = carla.Location(1.6,0,1.7)
		#cam_rotation = carla.Rotation(0,180,0)
		rgb_cam_transform = carla.Transform(rgb_cam_location)
		rgb_ego_cam = world.spawn_actor(rgb_cam_bp,rgb_cam_transform,attach_to=ego_vehicle, attachment_type=carla.AttachmentType.Rigid)
		#rgb_ego_cam.listen(lambda image: image.save_to_disk('/home/tushar/work/carlaData/Town02/rgb/%.6d.jpg' % image.frame))
		rgb_ego_cam.listen(lambda image: print(image))
		
		# --------------
		# Add a Depth camera sensor to ego vehicle. 
		# --------------
		
		depth_cam_bp = None
		depth_cam_bp = world.get_blueprint_library().find('sensor.camera.depth')
		depth_cam_bp.set_attribute("image_size_x",str(960))
		depth_cam_bp.set_attribute("image_size_y",str(540))
		depth_cam_bp.set_attribute("fov",str(105))
		depth_cam_location = carla.Location(1.6,0,1.7)
		#cam_rotation = carla.Rotation(0,180,0)
		depth_cam_transform = carla.Transform(depth_cam_location)
		depth_ego_cam = world.spawn_actor(depth_cam_bp,depth_cam_transform,attach_to=ego_vehicle, attachment_type=carla.AttachmentType.Rigid)
		depth_ego_cam.listen(lambda image: image.save_to_disk('/home/tushar/work/carlaData/Town02/depth/%.6d.jpg' % image.frame, carla.ColorConverter.LogarithmicDepth))

		# --------------
		# Add a Segentation camera sensor to ego vehicle. 
		# --------------
		
		semseg_cam_bp = None
		semseg_cam_bp = world.get_blueprint_library().find('sensor.camera.semantic_segmentation')
		semseg_cam_bp.set_attribute("image_size_x",str(960))
		semseg_cam_bp.set_attribute("image_size_y",str(540))
		semseg_cam_bp.set_attribute("fov",str(105))
		semseg_cam_location = carla.Location(1.6,0,1.7)
		#cam_rotation = carla.Rotation(0,180,0)
		semseg_cam_transform = carla.Transform(semseg_cam_location)
		semseg_ego_cam = world.spawn_actor(semseg_cam_bp,semseg_cam_transform,attach_to=ego_vehicle, attachment_type=carla.AttachmentType.Rigid)
		semseg_ego_cam.listen(lambda image: image.save_to_disk('/home/tushar/work/carlaData/Town02/semseg/%.6d.jpg' % image.frame, carla.ColorConverter.CityScapesPalette))

		
		# --------------
		# Place spectator on ego spawning
		# --------------
		"""
		spectator = world.get_spectator()
		world_snapshot = world.wait_for_tick() 
		spectator.set_transform(ego_vehicle.get_transform())
		"""

		# --------------
		# Enable autopilot for ego vehicle
		# --------------

		ego_vehicle.set_autopilot(True)
		# --------------
		# Game loop. Prevents the script from finishing.
		# --------------
		while True:
			world_snapshot = world.wait_for_tick()

	finally:
		# --------------
		# Stop recording and destroy actors
		# --------------
		client.stop_recorder()
		if ego_vehicle is not None:
			if ego_cam is not None:
				ego_cam.stop()
				ego_cam.destroy()
			if ego_col is not None:
				ego_col.stop()
				ego_col.destroy()
			if ego_lane is not None:
				ego_lane.stop()
				ego_lane.destroy()
			if ego_obs is not None:
				ego_obs.stop()
				ego_obs.destroy()
			if ego_gnss is not None:
				ego_gnss.stop()
				ego_gnss.destroy()
			if ego_imu is not None:
				ego_imu.stop()
				ego_imu.destroy()
			ego_vehicle.destroy()

if __name__ == '__main__':

	try:
		main()
	except KeyboardInterrupt:
		pass
	finally:
		print('\nDone with tutorial_ego.')
```
Saving each image is creating lots of problems and I want to store multiple images in numpy array (different numpy array for different camera sensor). I have been looking as how I can get the values and store them in an empty numpy array one by one, but I can't find anything for version 0.9.9.4. I just need large amount of training data and it's important that images are stored in the same sequence that they are generated.

I hope someone can help me with this.

Thanks

germanros1987, 2020-07-24:
@sergi-e I am pretty sure you could give good advise here.

khojir, 2020-07-30:
Hi,
each sensor has a `listen` function. according to your code, you are printing the received data. Instead of printing them, you can append the data to a `list` or a `numpy array`. When you have multiple sensors, you need to define multiple arrays or lists. 
Maybe you face situations like storing 2 depth data and 1 semantic segmentation. I'm not sure all sensors are working synchronized. But I think this isn't a big problem.

In this example, I'm using `list`. but you can also use `np.append(old, new)`:
```
depth_frames = []
depth_ego_cam.listen(lambda image: my_appender(image))

def my_appender(image):
   // do what ever you want
   global depth_frames
   depth_frames.append(image)
```

closed 2020-10-10 by stale[bot]