carla-simulator/carla #3095: Potential bug in the multi-stream sensor data transmission.

wx9698, 2020-07-22:
Hi team, 

Recently I was playing with the multi-stream sensor data transmission feature and encountered the std::bad_weak_ptr exception. I guess it might be a bug caused by the shared_from_this() function. Here are the re-production scripts.

script1.py    # This file is intended to spawn two sensors (lidar + camera)
``` python
import carla
import time

def do_something(data):
    if isinstance(data, carla.LidarMeasurement):
        print('lidar data frame:', data.frame)
    else:
        print('camera data frame:', data.frame)

def main():
    try:
        client = carla.Client('localhost', 2000)
        client.set_timeout(10.0)
        world = client.get_world()

        # set synchronous mode
        previous_settings = world.get_settings()
        world.apply_settings(carla.WorldSettings(
            synchronous_mode=True,
            fixed_delta_seconds=1.0 / 30.0))

        spawn_point = world.get_map().get_spawn_points()[0]
        blueprint = world.get_blueprint_library().filter("vehicle.*")[0]

        actor = world.spawn_actor(blueprint, spawn_point)

        blueprint_camera = world.get_blueprint_library().find('sensor.camera.rgb')
        blueprint_camera.set_attribute('image_size_x', '640')
        blueprint_camera.set_attribute('image_size_y', '480')
        blueprint_camera.set_attribute('fov', '110')
        blueprint_camera.set_attribute('sensor_tick', '0.1')
        transform_camera = carla.Transform(carla.Location(y=+3.0, z=5.0))
        camera = world.spawn_actor(blueprint_camera, transform_camera, attach_to=actor)
        camera.listen(lambda data: do_something(data))

        blueprint_lidar = world.get_blueprint_library().find('sensor.lidar.ray_cast')
        blueprint_lidar.set_attribute('range', '30')
        blueprint_lidar.set_attribute('rotation_frequency', '10')
        blueprint_lidar.set_attribute('channels', '32')
        blueprint_lidar.set_attribute('lower_fov', '-30')
        blueprint_lidar.set_attribute('upper_fov', '30')
        blueprint_lidar.set_attribute('points_per_second', '56000')
        transform_lidar = carla.Transform(carla.Location(x=0.0, z=5.0))
        lidar = world.spawn_actor(blueprint_lidar, transform_lidar, attach_to=actor)
        lidar.listen(lambda data: do_something(data))

        while True:
            world.tick()
            time.sleep(0.1)

    finally:
        if previous_settings:
            world.apply_settings(previous_settings)
        if actor:
            actor.destroy()
        if camera:
            camera.stop()
            camera.destroy()
        if lidar:
            lidar.stop()
            lidar.destroy()

if __name__ == "__main__":
    main()
```

script2.py # This file is intended to listen to the "alive" sensor and listen to it. This script will stop listen the sensor when the sensor is no longer alive.
```python
import carla

def do_something(data):
    if isinstance(data, carla.LidarMeasurement):
        print('lidar data frame:', data.frame)
    else:
        print('camera data frame:', data.frame)

def main():
    try:
        client = carla.Client('localhost', 2000)
        client.set_timeout(10.0)
        world = client.get_world()

        sensors = dict()

        while True:
            snapshots = world.wait_for_tick(30)
            ids = set()
            for snapshot in snapshots:
                ids.add(snapshot.id)
                if snapshot.id in sensors.keys():
                    continue
                actor = world.get_actor(snapshot.id)
                if actor.type_id.startswith('sensor'):
                    print('find sensor')
                    actor.listen(lambda data: do_something(data))
                    sensors[snapshot.id] = actor

            to_delete_sensor = []
            for sensor_id in sensors.keys():
                if not sensor_id in ids:
                    print('stop listening to dead sensor')
                    to_delete_sensor.append(sensor_id)
                    sensors[sensor_id].stop()
            for sensor_id in to_delete_sensor:
                sensors.pop(sensor_id)
                                
    except Exception as e:
        print(e)

if __name__ == "__main__":
    main()
```

Now we launch the carla simulator, script1 and script2. Then keep terminating script1 (which means keeping destroying sensors). Then the std::bad_weak_ptr might be thrown. Since there is a race condition, error might not be there every time when you kill the script1. You might need to try several times to get the error. Here is the output of the result.

script2.py
```
lidar data frame: 2175
lidar data frame: 2176
lidar data frame: 2177
camera data frame: 2176
lidar data frame: 2178
stop listening to dead sensor
terminate called after throwing an instance of 'std::bad_weak_ptr'
  what():  bad_weak_ptr
Aborted (core dumped)
```

After some investigations, I found that the error might come from this file - [LibCarla/source/carla/streaming/detail/tcp/Client.cpp - line 158](https://github.com/carla-simulator/carla/blob/master/LibCarla/source/carla/streaming/detail/tcp/Client.cpp#L158). The shared_from_this() function is called when all shared_ptrs of this class have been destructed and this callback is a dangling callback. 

Please let me know if it is correct and is there anything the team can do to fix this problem.

Best,
Minjun

germanros1987, 2020-07-24:
@bernatx this is very relevant for the current release. Please, have a look at it.

bernatx, 2020-07-24:
Hi Minjun Xu, glad you are back.
Thanks for all the details, I'm looking at it.

closed 2020-07-31 by bernatx

wx9698, 2020-08-08:
Hi team,

I was testing again with Carla 0.9.10. However, everytime when I terminated script1.py with ctrl+c, the carla server will get segmentation fault. Here is the output.
```bash
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=118144 LargeMemoryPoolOffset=249280 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

PR #3146 (2020-07-31): Update RPClib with fix of crash when closing session
#### Description

Fix a crash when closing sessions in the RPClib. We use a new version 2.2.1.c3 of the RPClib fork.

#### Where has this been tested?

  * **Platform(s):** Ubuntu, Windows
  * **Python version(s):** 2.7, 3.7
  * **Unreal Engine version(s):** 4.24

#### Fixes: #3095

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3146)