carla-simulator/carla #3097: How to add new light source on top of the vehicle?

YunzeMan, 2020-07-22:
Hi,

I am currently work on adding new source light on top of the vehicle. For example a vehicle roof-mounted light.

I wonder is it possible to customize it in python API or i need to do it in Unreal engine?

I've been working on it in Unreal for a while. But I don't know if this is the right direction to go.

Thanks!

germanros1987, 2020-07-24:
You need to do it from Unreal Engine. @XGodina could you help with this?

YunzeMan, 2020-07-24:
Thanks for your reply! @germanros1987 
I did see the document [https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/](https://carla.readthedocs.io/en/latest/tuto_A_add_vehicle/), and I followed it to create a car model.

However, since I'm adding a new top-mounted light, and I want to be able to toggle it just like other light sources, I wonder what is the right way to do this. Do I have to somehow bound this light to the reserved light instance carla.VehicleLightState.Special1? Thanks

germanros1987, 2020-07-24:
@Madecu you have worked on this. Could you help?

Madecu, 2020-07-24:
Hi. You need to add a light inside the blueprints car and name it in the "Variable name" slot as 'special1-X'

[image]

closed 2020-09-30 by YunzeMan