carla-simulator/carla #3099: ModuleNotFoundError: No module named 'libad_rss_python3'

vishal-246, 2020-07-23:
Hi, 

I've built RSS variant of carla and with python3 i have the libad_rss_python3 import error. error is shown below

germanros1987, 2020-07-24:
@fpasch could you please help with this?

fpasch, 2020-07-27:
That error looks like you're using a Python Egg without RSS enabled. Please ensure, that
1. you've build it with rss enabled (e.g. `make PythonAPI.rss.2`)
2. the search order in your PYTHONPATH is pointing to the right egg.

vishal-246, 2020-08-03:
Hi @fpasch  Sorry for the late replay.

A1. Yes i have built it with rss enabled only difference is i havent mentioned version like u have suggested. B'coz since `make PythoonAPI.rss` it self takes care of generating egg for python versions.

A2. For Python2 API its pointing to the right path as well as for Python3 if not it would have thrown the carla NoModule Found error.

closed 2020-10-11 by stale[bot]