carla-simulator/carla #3100: Carla cpp client - RPC

meitalcohenyaacobi, 2020-07-23:
Hi ,
We would like to run a self-implemented client for connecting to the simulator server and receive sensory data from the simulator. The client will run on remote machine and will connect to CARLA via RPC calls. 

As far as I understand from the examples: 
1. The RPC server uses rpclib, which uses MsgPack for serialization (https://github.com/carla-simulator/rpclib , https://github.com/carla-simulator/carla/issues/555)
2. LibCarla ([LibCarla\source\carla\rpc](http://carla.org/Doxygen/html/dir_83201f30ace02cdf218980110872f8a7.html))

what are the differences ?
what should we use?

Thank you.

germanros1987, 2020-07-24:
@nsubiron @bernatx could you comment on this?

bernatx, 2020-07-24:
Hi,
CARLA uses the RPCLib library for all the RPC commands.
The RPC folder in LibCarla is just the serialization of all commands that CARLA uses. There, we use the RPCLib for the serialization of our commands with MsgPack.
So, there is only one RPC library to use.

meitalcohenyaacobi, 2020-07-26:
Thanks a lot :)

closed 2020-10-04 by stale[bot]