carla-simulator/carla #3104: Question about implementing vehicles with Pacejka tire model in Carla

yyab, 2020-07-23:
Hi!
I'm planning to add a new car into Carla with more realistic vehicle lateral dynamical model, namely a "bicycle model" with Pacejka tire model. Though Carla has provided some tuning parameters on vehicle dynamics like tire friction, wheel damping rate, etc, the actual vehiclular model seems to be based on the Unreal engine vehicle blueprint, whose model is tied to the PhsicX engine. Is there any way to quickly swap that vehicle model? Do I have to make a custom blueprint class in the Unreal engine from scratch, or is there any quick way to do so? Thank you very much!

germanros1987, 2020-07-24:
@DSantosO @doterop could you comment on this?

whuxhzhang, 2020-07-28:
I have the same question,

bulus92, 2020-09-29:
I have the same question as well. Has there been any response to this question ?

DSantosO, 2020-10-01:
Hello @yyab, @whuxhzhang and @bulus92,
Sorry for the late response. In principle you can do it directly in BP in Unreal but with this you would loose most of the functionality that we provide for vehicles. In you look into the class hierarchy of vehicles, you would see that Carla's vehicles all inherit from ACarlaWheeledVehicle that inherit from AWheeledVehicle (from VehiclePhysX). Maybe you could create a Bicycle class that inherits from ACarlaWheeledVehicle and override the behaviour you want to replace. And then based your BP in that class.
Be aware that most of the Vehicle behaviour is controlled on ACarlaWheeledVehicle::ApplyVehiclePhysicsControl, if you replace this function to control your car instead of using UWheeledVehicleMovementComponent, I think it should be ok.

closed 2020-12-25 by stale[bot]