carla-simulator/carla #3106: LibCarla in C++ client

aviadziv, 2020-07-24:
I would like to integrate LibCarla in C++ project.
I have seen this: https://github.com/carla-simulator/carla/issues/810
And there is an exmple here:
https://github.com/carla-simulator/carla/tree/master/Examples/CppClient

So it it now ready for integrate in C++ project?

Another thing, it is said that LibCarla only depends on boost and rpcLib.
So if I build my own client that use LibCarla, is there any code there that prevent me to run this client on Android?

Thanks

germanros1987, 2020-07-24:
No blockers in principle, but we have never tried out. So please, be our guest. 
@bernatx do you anticipate any problems compiling libcarla for android?

closed 2020-10-04 by stale[bot]

jemishGhoghari, 2024-03-29:
Hi @aviadziv,

Are you able to somehow build Carla C++ Client on Windows? Is it possible to import LibCarla via CMake?

Actually, I want to include Carla C++ client headers in a Project that creates a bridge between Carla and Vehicle Control Software. 
I am new to Carla C++ API. However, I have used PythonAPI before.

I appriciate any help! 

Thanks!