carla-simulator/carla #3110: Is it possible to run CARLA server within the client side code?

AIasd, 2020-07-27:
Hi,

I am trying to run a CARLA server at some point within the client side code as the following:
```
cmd = shlex.split('perl CarlaUE4.sh -opengl -carla-rpc-port=2000 -carla-streaming-port=0')
subprocess.Popen(cmd)
```
After these two lines, I observe by using `top` in the terminal that CarlaUE4 is running successfully at localhost:2000. However, I find that the client fails to connect to the server no matter how long they try:
```
while True:
    try:
        client = carla.Client('localhost', 2000)
        client.load_world(town_name)
    except:
        print('failed to connect to server')
```

If instead, I run the server externally and run the client code afterwards, they work normally.
I wonder what's wrong with my code on running the CARLA server within the client side code? Is this doable?

Thanks!

germanros1987, 2020-07-27:
Hi,

I think your question needs to be rephrased. You mean if it is possible to launch the CARLA server from Python or any other language. That shouldn't be a problem. I have done it for several projects in Python.

Now, why are you invoking CarlaUE4.sh with a perl interpreter?

AIasd, 2020-07-28:
@germanros1987 Thank you for the quick response!
It turns out that if I add time.sleep(5) after launch the CARLA server from Python before I run 
```
client = carla.Client('localhost', 2000)
client.load_world(town_name)
```
will make it work.

closed 2020-07-28 by AIasd