carla-simulator/carla #3117: Custom quality level

thytless, 2020-07-28:
Hello! I am working with carla 0.9.9 which runs as a docker container, and I am looking for something other than cmdline option **'-quality-level'** that customizes rendering in order to ease pressure of rendering thread and get a better fps in sync mode.

I've read about CarlaSettings.cpp, which looks like taking quality-level as input and update the server settings. I've also tried to modify some *.ini config under ./.config/Epic/ and ./Engine/Config/ (modify `sg.ViewDistanceQuality`), but it did not work.

Do we have other access to these options?

thytless, 2020-07-28:
BTW, I am not sure if it is the case that rendering not fast enough leads to low fps (about 35), how could I test it? I just found that one thread of Carla  always takes 95%+ cpu in htop gui.
My cpu is Intel E5 2680 and gpu is 1080Ti

germanros1987, 2020-07-30:
Modifying the .ini file should have an effect in the rendering capabilities. @marcgpuig could you comment on this?

marcgpuig, 2020-08-18:
@thytless
Yes, you should be able to customize these attributes but with caution, most of them are carefully set to specific values.
Make sure you are correctly modifying them, and make sure to understand what are they doing in the game since most of them are not easily appreciable.

closed 2020-08-20 by thytless

licj15, 2020-12-18:
Hello everyone,

Thanks for the insightful discussion!

I am new to the CarlaUE4 Engine.
Could you share what you changed in the *.ini file and how will it affect the FPS? Or do you have any references to recommend to someone who is new to it?

Thanks!