carla-simulator/carla #3119: Error when using get_waypoint() with carla.LaneType.Any in Windows

LikhitaNukala, 2020-07-28:
I have executed the following command for OpenSCENARIO.
**python scenario_runner.py --openscenario L:\carla_latest\scenario_runner-0.9.9\srunner\examples\CyclistCrossing.xosc**

I'm getting the following.

WARNING:OpenScenarioConfiguration:Wrong OpenDRIVE map in use. Forcing reload of CARLA world
Preparing scenario: CARLA:CyclistCrossing
ScenarioManager: Running scenario OpenScenario
Traceback (most recent call last):
  File "scenario_runner.py", line 382, in _load_and_run_scenario
    self.manager.run_scenario()
  File "L:\carla_latest\scenario_runner-0.9.9\srunner\scenariomanager\scenario_manager.py", line 216, in run_scenario
    self._tick_scenario(timestamp)
  File "L:\carla_latest\scenario_runner-0.9.9\srunner\scenariomanager\scenario_manager.py", line 260, in _tick_scenario
    self.scenario_tree.tick_once()
  File "C:\Users\lnukala\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\py_trees\behaviour.py", line 158, in tick_once
    for unused in self.tick():
  File "C:\Users\lnukala\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\py_trees\composites.py", line 578, in tick
    for node in child.tick():
  File "C:\Users\lnukala\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\py_trees\composites.py", line 578, in tick
    for node in child.tick():
  File "C:\Users\lnukala\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\py_trees\composites.py", line 578, in tick
    for node in child.tick():
  File "C:\Users\lnukala\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.7_qbz5n2kfra8p0\LocalCache\local-packages\Python37\site-packages\py_trees\behaviour.py", line 249, in tick
    new_status = self.update()
  File "L:\carla_latest\scenario_runner-0.9.9\srunner\scenariomanager\scenarioatomics\atomic_criteria.py", line 648, in update
    current_wp = self._map.get_waypoint(current_loc, lane_type=carla.LaneType.Any)
OverflowError: can't convert negative value to unsigned int
can't convert negative value to unsigned int
Destroying ego vehicle 258
No more scenarios .... Exiting

**Desktop **
 - OS: Windows 10
 - CARLA Version CARLA 0.9.9
 - Python version 3.7.6
 - Version  0.9.9

glopezdiest, 2020-07-28:
Adding @bernatx and transferring it to the CARLA main repository, as this doesn't seem to be ScenarioRunner related anymore

fabianoboril, 2020-10-15:
This still seems to be relevant. According to discord, people still see this issue in CARLA 0.9.10

bernatx, 2020-10-16:
I'm working on this issue

closed 2020-10-19 by bernatx

zhengsongyue, 2020-10-21:
[image]
Hello, have you fixed this issue? seems like i met the same issue on Ubuntu 18.04.

glopezdiest, 2020-10-21:
Hey @zhengsongyue, you seem to not be finding the *carla/PythonAPI/carla/agents* folder. Can you check that you have the *carla/PythonAPI/carla* folder added to your PYTHONPATH?

Also, why didn't you create a new issue instead of posting it here? Your error is unrelated to the one above.

TeeManiac, 2020-10-29:
I have been working with xml scenario files and everything was working fine. When I tried to run OpenSCENARIO files, only CatalogExample, ChangingWeather and OscControllerExample was working and I am getting the following error when trying to run CyclistCrossing, PedestrianCrossingFront, FollowLeadingVehicle and LaneChangeSimple files.

"current_wp = self._map.get_waypoint(current_loc, lane_type=carla.LaneType.Any)                                                                      OverflowError: can't convert negative value to unsigned int                                                                                             
can't convert negative value to unsigned int                                                                                                            
Destroying ego vehicle 5752                                                                                                                             
ERROR: failed to destroy actor 5752 : unable to destroy actor: not found                                                                                
No more scenarios .... Exiting

**Desktop **

OS: [Windows 10]
CARLA Version [CARLA 0.9.10]
Python version [3.9.0]
Scenario_runner Version [0.9.10]

I've seen that this issue is for Windows. Please help me to resolve this issue.

[image]

glopezdiest, 2020-10-29:
hey @TeeManiac. That's not an ScenarioRunner error but a CARLA one. We recently fixed it with PR's https://github.com/carla-simulator/carla/commit/aedd70911905d1cb80844c55cfc324469de07c90 and https://github.com/carla-simulator/carla/commit/530f56e78b999550670c7e5002b9b740218b8a58.

If you are using CARLA from source I'd recommend either cherrypicking those two commit, or upgrading to current *dev* branch

TeeManiac, 2020-11-03:
@glopezdiest Thank you..it worked now

aakashsehgal, 2021-04-22:
@glopezdiest i am facing the same issue on Win10 and Carla 0.9.10 (Binary)...any idea how could this be resolved ?

glopezdiest, 2021-04-22:
Unfortunately, you can't change any code of the package so you'd have to either:
1) Upgrade to 0.9.11
2) Get CARLA 0.9.10 from source, apply the PR's and make a new package
I'd like to give you a package with those changes but `Win10 and Carla 0.9.10 (Binary)` is a pretty bad configuration for me to do it on. I'll see if I can do it but I can't promise anything

aakashsehgal, 2021-04-23:
Upgrading to 0.9.11 solved the problem. Thanks.