carla-simulator/carla #3122: bad_weak_ptr

guntherkrehl, 2020-07-28:
(This is a copy of the forum entry: https://forum.carla.org/t/carla-client-core-throwing-bad-weak-ptr/379)

Hi everyone,

Ever since I started using CARLA i used the ROS bridge in conjunction with it. I love all of the stuff you’ve created and it works like a charm, almost. One error I have been encountering since 0.9.6 (when I started) was that when I remove actors (most commonly ego actor with lots of sensors) there is roughly a 50/50 chance that the simulator throws with a “bad_weak_ptr” error.
I hoped this would be resolved over time, since I also saw others commenting on it, but no luck. I still have these issues, I’ve tried to work it out myself but is seems to have something to do with your core stream/shared_pointer lifecycle handling, and I didn’t want to mess with that code :wink:

I am working with Linux, tried multiple systems, everywhere the same result. I have now implemented my own C++ ros bridge for Carla, and am experiencing the exact same issue like the python bridge. This hints towards it being a general core library issue.

Here is the immediate very cryptic gdb result:

terminate called after throwing an instance of ‘std::bad_weak_ptr’
what(): bad_weak_ptr

.....

The full error codes can be found in the forum post: https://forum.carla.org/t/carla-client-core-throwing-bad-weak-ptr/379

.....

Thank you very much and keep on with the great work you’re doing!

Best,
Gunther

germanros1987, 2020-07-30:
@guntherkrehl you are always too nice to us...undeserved!  (but thanks).

Let's take care of this as soon as we can. @bernatx @joel-mb @Axel1092 could you please coordinate a solution for this bug? We want to make ROS a 1st class citizen.

guntherkrehl, 2020-08-20:
Hi Carla team,

Will this issue be worked on before the 0.9.10 release?
Also, I would like to clarify, that this is not a ROS related issue. Generally any client that deals with spawning/removing/managing actors is affected by this (very critical). I mentioned the ROS bridge, because it's easily reproducible using that tool by creating an ego vehicle with lots of sensors.

Best,
Gunther

germanros1987, 2020-08-20:
@marcgpuig could you please track this issue?

PR #3295 (2020-09-16): Fixed bad_weak_ptr

#### Description
- Fixed the `bad_weak_ptr` descrived in #3122
- Updated several `boost::asio` calls that are deprecated in boost 1.72.0

Fixes #3122

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04
  * **Python version(s):** 2 and 3
  * **Unreal Engine version(s):** 4.24

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/carla/3295)
closed 2020-09-16 by bernatx