carla-simulator/carla #3123: Adding new blueprint types to the Actor Registry?

VictorWeidar, 2020-07-28:
Hello!

For a while now I have been trying to register new actors to Carla, so that I may access them in our python scripts. After little bit of reverse engineering I managed to figure out how to register classes that were already included in UE4 such as Directional lights or Atmospheric Fog. I did this using the "ActorView.h" and "ActorRegistry.cpp" scripts.

However, when I try to add my own blueprint classes, I just get an error that says "Use of undeclared Identifier" when I try to start the engine. I saw that you had added a "TrafficSignBase" blueprint that derives from AActor and that works fine, however when I try to add my "CustomBlueprintBase" that also derives from AActor I get the error mentioned above.

The error mentioned occurs in the "ActorRegistry.cpp" script first in the part where you cast to the actor type itself. As mentioned this works perfectly when adding things that are included in UE4, but now with my own items.

Do you have any tips that could help me include our own blueprint classes for later use in python? Am I going about the right way to register my own blueprint classes or am I missing a step?

Thanks

germanros1987, 2020-07-30:
@marcgpuig @Madecu could you please provide some advice on what is the easiest path to do this?

closed 2020-10-10 by stale[bot]

VictorWeidar, 2020-11-09:
Hello I'm not sure If i should post a new issue, but if so please let me know.

This question has become relevant for me again so I would appreciate some assistance with this yet again :)

@germanros1987 @marcgpuig @Madecu