carla-simulator/carla #3124: Exception while installing dependencies in Ubuntu 18.04

Jaskaran-Singh-Sodhi, 2020-07-28:
I'm trying to build CARLA in Ubuntu 18.04 by following the documentation.

While entering the following commands, an exception occurs,

sudo apt-get install build-essential clang-8 lld-8 g++-7 cmake ninja-build libvulkan1 python python-pip python-dev python3-dev python3-pip libpng-dev libtiff5-dev libjpeg-dev tzdata sed curl unzip autoconf libtool rsync libxml2-dev &&
pip2 install --user setuptools &&
pip3 install --user setuptools &&
pip2 install --user distro &&
pip3 install --user distro

[image]

Please help. Thanks in advance!

gsujansai, 2020-07-29:
Hey. Could be related to [link](https://askubuntu.com/questions/1040392/easy-install-command-not-found). But this seems to python issue and you ll probably have better luck in finding an answer on stackoverflow.

germanros1987, 2020-07-30:
Agreed. Please try StackOverflow for this particular issue.

closed 2020-07-30 by germanros1987