carla-simulator/carla #3126: Version mismatch, Simulator API version: xxx - dirty

Privilger, 2020-07-29:
Hi,

I meet a strange problem.

I have a Windows with Carla server 0.9.9 on it. Python version: 3.5.3
And another Ubuntu 16 with Carla client 0.9.9 on it. Python version: 3.5.6

When I connect the client with the server, it said that:

```
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 

WARNING: Client API version     = 0.9.9 

WARNING: Simulator API version  = 0.9.9-dirty 
```
What is the meaning of **dirty** ?

When I using the `manual_control.py`, the interface is very stuck, I think the fps might be 3 or 4.

Another problem is that, it works well at the first beginning actually. 

After I `make rebuild` it, the problem come up.

thytless, 2020-07-29:
1. Dirty means you build the Python API with a dirty git repo (with uncommited changes)
2. (Stuck...) Maybe you should post your GPU and CPU info?

Privilger, 2020-07-29:
> (Well, I'm not developer)
> 
>     1. Dirty means you build the Python API with a dirty git repo (with uncommitted changes)
> 
>     2. (Stuck...) Maybe you should post your GPU and CPU info?

Hi, 

Thanks for your reply. The first answer does solve my problem.

The Stuck is quite strange... it works well on my computer A, but stuck on my computer B.

germanros1987, 2020-07-30:
Thanks @thytless for helping!

closed 2020-07-30 by germanros1987