carla-simulator/carla #3127: White image with default attributes

fpasch, 2020-07-29:
Setting the default values (as described in the documentation) leads to a completely white camera image.,

                        bp.set_attribute('shutter_speed', '60.0')
                        bp.set_attribute('iso', '1200.0')
                        bp.set_attribute('fstop', '1.4')
                        bp.set_attribute('min_fstop', '1.2')
                        bp.set_attribute('exposure_mode', 'manual')
                        bp.set_attribute('exposure_compensation', '3.0')
                        bp.set_attribute('exposure_min_bright', '0.1')
                        bp.set_attribute('exposure_max_bright', '0.3')

germanros1987, 2020-07-30:
@marcgpuig --master of sensors, ruler of blueprints, breaker of shaders-- do you care to comment?

fpasch, 2020-07-30:
solved by #3115

closed 2020-07-30 by fpasch