carla-simulator/carla #3130: Lens distortion attributes

ffabi, 2020-07-29:
Can you help me find the correlation between the lens distortion attributes defined in Carla and radial/tangential distortion coefficients?

OpenCV documentation defines k1-k6 for radial distortion and p1, p2 for tangential distortion:
https://docs.opencv.org/2.4/modules/calib3d/doc/camera_calibration_and_3d_reconstruction.html

I would like to define these values, can you help me calculate the attributes needed for Carla camera sensor?
Can you provide a more detailed description of these values?

Maybe lens_k = k1 and lens_kcube = k2 ?

I could not find these Carla attributes in Unreal engine documentation either.

germanros1987, 2020-07-30:
@marcgpuig which model did you follow to implement the lens distortion?

ffabi, 2020-07-30:
For a while, pinhole camera model was sufficient for me, but I would like to generate images with a given radial and tangential distortion according to the Brown–Conrady model. I would like to mimic a real-world sensor which has defined k1, k2, p1, p2, k3 parameters.

germanros1987, 2020-07-30:
@marcgpuig are we using the Brown–Conrady model to create the lens distortion?

jerryliu20001, 2020-08-27:
I have the same problem about the lens distortion attributes defined in CARLA.  
If the default value of the **lens_k** and **lens_kcube** mean without distortion, the assumption of the **lens_k=k1** and **lens_kcube=k2** are unreasonable.  
In addition, I am also confused about the other attributes.  

[image]

germanros1987, 2021-01-26:
@marcgpuig do you have 10 min to answer this?

YifuTao, 2021-02-17:
Hi, I have found that the distortion model is "`plumb_bob`".

I used the ros-bridge, ran `rostopic echo /carla/ego_vehicle/rgb_view/camera_info` and get something like this:

>  ---
> header: 
>   seq: 10
>   stamp: 
>     secs: 435
>     nsecs: 313154742
>   frame_id: "ego_vehicle/rgb_test"
> height: 480
> width: 848
> distortion_model: "plumb_bob"
> D: [0.0, 0.0, 0.0, 0.0, 0.0]
> K: [431.4655346201811, 0.0, 424.0, 0.0, 431.4655346201811, 240.0, 0.0, 0.0, 1.0]
> R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
> P: [431.4655346201811, 0.0, 424.0, 0.0, 0.0, 431.4655346201811, 240.0, 0.0, 0.0, 0.0, 1.0, 0.0]
> binning_x: 0
> binning_y: 0
> roi: 
>   x_offset: 0
>   y_offset: 0
>   height: 0
>   width: 0
>   do_rectify: False
> ---

We changed `K ` (and hence `P`) by setting the fov and image size. Specifically, K1, K5 (fx, fy) by fov (fov = 2 arctan (fx/2f)) and K3, K6 (cx, cy) by image size (halve). Definition of D,K,R,P and more information of plumb_bob distortion model can be found in http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/CameraInfo.html.

Question remained: we still cannot set `D`, the distortion model. I have tried to set the [CARLA Camera lens distortion attributes](https://carla.readthedocs.io/en/latest/ref_sensors/#rgb-camera) as mentioned above, but that does not lead to changes in nor D or other matrix in the above camera information.

Can you help on this? It would be very helpful for the community if the camera distortion section documentation can be improved.

Many thanks in advance!

Axel1092, 2021-03-03:
Hi all, we are on the way of changing the lens distortion algorithms as we found some issues and inaccurate results with the current implementation. We plan to expose these parameters so that they can be tuned easily.

jimmyw404, 2021-08-16:
> Hi all, we are on the way of changing the lens distortion algorithms as we found some issues and inaccurate results with the current implementation. We plan to expose these parameters so that they can be tuned easily.

Hi @Axel1092 , any update on this! I'm very interested in getting basic lens distortion with matching OpenCV parameters!

Kait0, 2021-11-04:
I am also interested in this issue.

AbanobSoliman, 2021-11-29:
Hi, I am also interested in this topic/issue. any updates ! Thanks

kwincu, 2022-01-25:
> Hi all, we are on the way of changing the lens distortion algorithms as we found some issues and inaccurate results with the current implementation. We plan to expose these parameters so that they can be tuned easily.

I'm very interested your method, please

jimmyw404, 2022-04-16:
I am still interested in this!

qiangt, 2022-05-16:
Does any one know any update on this? I just checked the latest Carla version 0.9.13, and it still uses the old model

yuanhaorannnnnn, 2022-08-03:
so upset to see that no updates with such a basic concept

1999010168, 2022-12-02:
also sad

dengping023, 2023-04-03:
Looking forward to distortion model of opencv used in carla

Hassan-BABAOUSMAIL, 2023-05-02:
> Hi all, we are on the way of changing the lens distortion algorithms as we found some issues and inaccurate results with the current implementation. We plan to expose these parameters so that they can be tuned easily.

Hello sir,
Is there any update? Which model was used in CARLA to model distortion? How can 
 we modify it?

ZijianHan, 2023-05-13:
any updates

> Hi all, we are on the way of changing the lens distortion algorithms as we found some issues and inaccurate results with the current implementation. We plan to expose these parameters so that they can be tuned easily.

any updates on the distortion model?

alanhuang1990, 2023-05-24:
It seems that carla use this distortion model: https://www.ssontech.com/content/lensalg.html

XinjieInformatik, 2023-06-25:
Any update for such an important property?

zhengxh-git, 2023-11-23:
I am still interested in this!

leoleo0714, 2024-03-01:
I am interested in this!

MinnanZhou, 2024-05-11:
Still interested in this issue

zmd-cc, 2025-07-23:
Still interested in this issue.
Hope Carla team could give more infos.