carla-simulator/carla #3135: Carla 0.9.9.4-Multiple Manual Controls not working - Independent Client-Server Configuration

chandeep11, 2020-07-30:
I'm running Carla server in one system and the Python Client in another system. Both are connected to same LAN Network.
I can able to run 1st manual control successfully in the client system. When trying to run 2nd manual control in the client(while 1 is still functional), I'm getting the following error. 

`RuntimeError: time-out of 2000ms while waiting for the simulator, make sure the simulator is ready and connected to 192.168.8.101:2000
  The server could not send the OpenDRIVE (.xodr) file:
  Make sure it exists, has the same name of your town, and is correct.
`

Similar behavior is observed in Custom binaries which are build from the source versions(0.9.7). What could be the possible cause of this issue?

germanros1987, 2020-07-31:
@chandeep11 does this behavior occur also in 0.9.9.4?

chandeep11, 2020-07-31:
@germanros1987  Yes, it occurs in 0.9.9.4 too.

germanros1987, 2020-07-31:
@bernatx could you take a look at this and report back?

bernatx, 2020-07-31:
We tested and we can use more than one manual control. We tested through internet mixing windows and linux and it works also, but sometimes we need to specify a greater timeout to let the client connect to the server. The 2 seconds by default are not enough in some situations, so I advice to set that time bigger to allow for busy networks or internet.

I think that was the problem happened.

closed 2020-10-11 by stale[bot]