carla-simulator/carla #3138: PythonAPI world.get_lightmanager wrong name in documentation

gsujansai, 2020-07-30:
Hi, 

The documentation for LightManager in PythonAPI has the function name as get_light_manager [link](https://carla.readthedocs.io/en/latest/python_api/#carla.World.get_light_manager).

But the function name is 
`.def("get_lightmanager", CONST_CALL_WITHOUT_GIL(cc::World, GetLightManager))` in [world.cpp](https://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/source/libcarla/World.cpp)

Will the documentation be updated?

germanros1987, 2020-07-31:
@sergi-e could you please take care of this?

sergi-e, 2020-08-31:
Hi, @gsujansai !

This issue was properly fixed. Thanks a lot for reporting it :)

closed 2020-08-31 by sergi-e

analog-cbarber, 2020-10-09:
This was only fixed for the python api doc for `World` there are still places in the documentation that refer to the incorrect name: https://github.com/carla-simulator/carla/search?q=get_light_manager

sergi-e, 2020-10-09:
Hi, @analog-cbarber !

Completely true. These instances have been fixed in this branch: https://github.com/carla-simulator/carla/tree/sergi-e/warning-docs-version and will be merged as soon as possible, along with other minor changes.