carla-simulator/carla #3145: No velocity or acceleration on log files

lucasnmatias, 2020-07-31:
I have been working with carla simulator but have facing some problems when dealing with the log files. When I try to use `get_velocity()` and `get_acceleration()` from live simulation it works fine, but when I use those two functions during replaying a log file it returns 0 only and I can't get those information from vehicles. Do you have any guess about it? There is any way to overcome this problem?

yanlai00, 2020-08-01:
Facing the same problem for 0.9.9 when enabling autopilot. It seems that once autopilot is enabled, even if you disable it later, get_velocity() still always return (0, 0, 0).

yanlai00, 2020-08-01:
There also seems to be some problem with get_location(). Not sure what kind of problem yet. However, when I enable and disable autopilot mode alternately with some probability the location of the car seems to update only once several frames.

glopezdiest, 2020-08-03:
Hey @lucasnmatias. Yes, when replaying a log file, the physics of all the vehicle are disabled, and the don't move, but teleport around, hence why their velocity is zero. @yanlai00, same thing happens for the traffic manager with hybrid mode, when active, all physics are disabled, and the vehicle teleport around.

We recently changed the recorder and its *show_recorder_file_info*. *start_recorder* now has a parameter that, if set to true, returns all the information about the simulation, not just the one needed to replay it. There isn't just one commit that adds it so I can't point that one out, but [here](https://github.com/carla-simulator/carla/commit/f39bc702422261865edf69354cc4b3119681e68e) is the one that adds the velocity.

yanlai00, 2020-08-03:
@glopezdiest Thanks for response! However, even if I set the attribute "role_name" of vehicle to "hero", which should disable hybrid mode on this vehicle, the problem still persists.

glopezdiest, 2020-08-03:
That depends, the hybrid mode has a radius around the ego vehicle  (the one with the *hero* tag). Are you checking the actors that are inside that radius, or those that are outside?

yanlai00, 2020-08-03:
I am checking exactly the ego vehicle.

glopezdiest, 2020-08-03:
Okay, I'll check your code in detail then, and report back

glopezdiest, 2020-08-03:
Indeed, that does seem to be the case. Tagging @jackbart94, he will know much more about the hybrid mode. Perhaps there's something we are missing

lucasnmatias, 2020-08-11:
> Hey @lucasnmatias. Yes, when replaying a log file, the physics of all the vehicle are disabled, and the don't move, but teleport around, hence why their velocity is zero. @yanlai00, same thing [...]

I see, another question, is this also supported on Carla 0.9.8?

glopezdiest, 2020-08-31:
No, the recent changes aren't part of any release yet, they are just in current master branch

raymanfx, 2021-03-22:
Bump. This still happens with Carla 0.9.11. Position data etc. is available during playback, but velocity is **not**, even if additional_data is `True` in `start_recorder`.

raymanfx, 2021-03-22:
Apparently commit https://github.com/carla-simulator/carla/commit/f39bc702422261865edf69354cc4b3119681e68e was supposed to add the velocity information (and more), but it does not appear to be working on Carla 0.9.11, the tagged release containing the commit.

glopezdiest, 2021-03-23:
Hey @raymanfx, it seems to be working fine for me in both 0.9.11 and current dev. I just created a script that:
1) starts recording: `client.start_recorder(path, True)`
2) spawns a vehicle and fixes its speed
3) waits for some seconds before stops recording
4) Gets the recorder info and saves it into a file `client.show_recorder_file_info(path, True)`
With this, I didn't see any problems, so how are you running the recorder?

raymanfx, 2021-03-24:
Hi @glopezdiest, thanks for taking a look.
This one is actually easy to reproduce with the official examples.

I modified `PythonAPI/examples/manual_control.py` and changed line 403 (in the 0.9.11 release) to `client.start_recorder("manual_recording.rec", True)` to record the additional metadata.

Using that script, I pressed CTRL+R to start the recording, drove for a bit, pressed CTRL+R to end the recording and finally CTRL+P to start the playback. The speed in the HUD is always 0 km/h, meaning that the velocity data is missing from the recording AFACIT.

glopezdiest, 2021-03-24:
Oh no, that's a different issue. The recorder has a somewhat dual functionality.

The first one (and original behavior) allows the user to reenact the same scene over and over again (using `client.replay_file(...)`).
The other one is used to get information of the scene (using `client.show_recorder_file_info()`).

The behavior you are seeing is the first one and it doesn't matter whether or not you start the recorder such that all the velocities are saved, because the playback just doesn't use that information. Instead, the way it works is by teleporting the actors through town, not by physically moving them.

raymanfx, 2021-03-24:
Thanks for clearing that up. I was not able to extract that kind of information from the recorder documentation.
Would it be possible to just "replay" the velocity values from the log while teleporting the actors around? It sounds like all the information is there and I think it would enhance the record/playback feature quite a bit.

glopezdiest, 2021-03-24:
> I was not able to extract that kind of information from the recorder documentation.

Yeah, the teleporting part is definitely not at the Docs. We're gonna have to update that.

> It sounds like all the information is there and I think it would enhance the record/playback feature quite a bit.

The playback was the original implementation of the recorder which is why it only uses the bare minimum information. It definitely would be an improvement but it would also need a complete redo to be able to use that information, and it isn't really a priority right now.

> Would it be possible to just "replay" the velocity values from the log while teleporting the actors around?

Not sure, but it isn't that simple, mainly because those values are gotten from the server, and I don't think there is a way to "fake" those values. The other alternative would be my previous comment, but again, that's even harder to do.

raymanfx, 2021-03-24:
I see. Sounds like I should just record the metadata when the agent + recorder is running and then sync it with the frames I record off the playback by matching the world snapshot timestamps.

Since this issue is about the information in the log files (which is present as you confirmed), I guess it can be closed.

closed 2021-03-25 by glopezdiest

ParthAD3, 2021-03-26:
Hey all 4 of you guys..

I can upon this issue looking to find a way how to get a plot the acceleration, velocity, and torque curve of my ego vehicle..

1. To confess i don't know how to use setters and getters, i am confused do i need to write my own script or do i have to run manul_control.py and then call getters..
Can someone from you guys post their code of setters and getters =  apply.throttle() getters = get.velocity() 

Also, i also tried to record and i have the log file but i don't know how to extract data from that test1.log file?