carla-simulator/carla #3150: Carla installation issue on Ubuntun laptop

carlaUE4new, 2020-07-31:
I am using the quick installation method to install Carla on my Ubuntu laptop, I followed the instructions 

sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 92635A407F7A020C
sudo add-apt-repository "deb [arch=amd64 trusted=yes] http://dist.carla.org/carla-0.9.9/ all main"

then run sudo apt-get install carla-simulator

but run into following errors (i did pip install --user pygame numpy),any suggestions?

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 carla-simulator : Depends: python-numpy but it is not going to be installed
                   Depends: python-pygame but it is not installable
E: Unable to correct problems, you have held broken packages.

germanros1987, 2020-08-19:
Sorry for the delay. It is vacation season. @joel-mb could you please take a look at this?

joel-mb, 2020-08-20:
Hi @carlaUE4new!
Are you still facing the issue? Are you using Ubuntu 18.04? You may want to use `aptitude` instead of `apt-get` to resolve the unmet dependencies. You can also try installing `python-numpy` and `python-pygame` manually before installing carla.
```sh
sudo apt-get install python-numpy python-pygame
```

carlaUE4new, 2020-08-20:
thank you. problem is fixed, I had some set up issue on env variable

joel-mb, 2020-08-20:
perfect! thanks for the feedback!

closed 2020-08-20 by joel-mb

CBalaji, 2021-01-01:
> Hi @carlaUE4new!
> Are you still facing the issue? Are you using Ubuntu 18.04? You may want to use `aptitude` instead of `apt-get` to resolve the unmet dependencies. You can also try installing `python-numpy` and `python-pygame` manually before installing carla.
> 
> ```shell
> sudo apt-get install python-numpy python-pygame
> ``` 

In my case that doesn't work as iam on ubuntu 20.10  i tried installing via pip and the issue persists,  

> sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla $(lsb_release -sc) main"
for the $(lsb_release -sc)  part i typed bionic manually  as neither groovy,focal,buster,stretch-- worked

CBalaji, 2021-01-01:
>@joel-mb 

>Hi@joel-mb

> 
> ```shell
> sudo apt-get install python-numpy python-pygame
> ``` 

In my case that doesn't work as iam on ubuntu 20.10  i tried installing via pip and the issue persists,  
`
`> sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla $(lsb_release -sc) `main"`
for the $(lsb_release -sc)  part i typed bionic manually  as neither groovy,focal,buster,stretch-- worked

joel-mb, 2021-01-07:
Hi @CBalaji,
Right now, the `apt` repository is only providing packages for `bionic` (i.e., Ubuntu 18.04). Moreover, all the packages we are providing are also targeting `bionic`. 
I've seen that there are some users that have managed to build CARLA on Ubuntu 20.04 from source. Please, take a look here: https://github.com/carla-simulator/carla/pull/2787) for more information.