carla-simulator/carla #3152: Is there a way to see the UE4 window on a remote server?

songyanghan, 2020-08-02:
I am running ./CarlaUE4.sh on a powerful remote server. I have tried X forwarding and VNC viewer, but they fail to show the window popped up. Is there a way to see the UE4 window from a local PC while running CARLA on a remote server?

AIasd, 2020-08-06:
also interested in knowing this!

zhukevin04, 2020-08-09:
> I am running ./CarlaUE4.sh on a powerful remote server. I have tried X forwarding and VNC viewer, but they fail to show the window popped up. Is there a way to see the UE4 window from a local PC while running CARLA on a remote server?

Hi, i am also trying that on server. When i launch Xorg, there is an error `(EE) Fatal server error (EE) no screens found`
Then i use P100 with CUDA10.2 and nvidia-smi works.

germanros1987, 2020-08-19:
I am not sure if the Xforwarding is going to work, but in any case it seems your problem is not with the X forwarding yet, but with the configuration of the Xorg server.

It is possible to configure CARLA to run in a headless server, but it takes a few steps.

* Make sure you have a Ubuntu 18.04 system
* Fetch the latest NVIDIA driver: wget http://download.nvidia.com/XFree86/Linux-x86_64/450.57/NVIDIA-Linux-x86_64-450.57.run
* Install the driver: sudo /bin/bash NVIDIA-Linux-x86_64-450.57.run --accept-license --no-questions --ui=none
* Install xserver related dependencies: sudo apt-get install -y xserver-xorg mesa-utils libvulkan1
* Configure xserver: sudo nvidia-xconfig -a --virtual=1280x1024
* Run xserver: sudo X :0 &
* Run CARLA: DISPLAY=:0.GPU_ID SDL_VIDEODRIVER=x11 ./CarlaUE4.sh -vulkan

Notes:
* Make sure that within the container the env variable SDL_VIDEODRIVER is set to x11, i.e., SDL_VIDEODRIVER=x11
* Make sure that within the container libvulkan1 is installed

That should launch the CARLA server in a headless node using vulkan. You can check that this is working by running any carla example.

Regarding the X forwarding, I am not sure if the X forwarding will forward the server window. In the worst case you can fetch the spectator camera (which is the camera the server shows) and display it locally using the CARLA API.

Hope this information helps.

songyanghan, 2020-08-24:
Thanks all. I find that [carlaviz](https://carla.readthedocs.io/en/latest/plugins_carlaviz/) is a good solution.

closed 2020-08-24 by songyanghan