carla-simulator/carla #3153: ModuleNotFoundError: No module named 'carla'

songyanghan, 2020-08-02:
OS: Ubuntu 18.04
Carla: 0.9.9

Hi,

I just downloaded carla recently. I encountered the error ModuleNotFoundError: No module named 'carla' when trying to execute the command python spawn_npc.py as suggested by the document. My carla-simulator is installed under "/opt/carla-simulator/". Under "/opt/carla-simulator/PythonAPI/carla/dist", I see the file "carla-0.9.9-py3.7-linux-x86.64.egg". In the file "spawn_npc.py", I see "sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg'% ". It is correct and supposed to work. Where is the error?

whuxhzhang, 2020-08-04:
Have you configured the PythonPath environment ？

XieJinLiang, 2020-08-04:
if you have install python 3.7，maybe you should use: python3 spawn_npc.py

songyanghan, 2020-08-04:
I change to a conda environment with python=3.7. It works now. Thanks.

closed 2020-08-04 by songyanghan