carla-simulator/carla #3154: cd PythonAPI/examples && python3 spawn_npc.py issue

gengtsh, 2020-08-03:
when i follow the linux build instruction: 
make PythonAPI && cd PythonAPI/examples && python3 spawn_npc.py
i encounter the following issues: 
python3 spawn_npc.py

destroying 0 vehicles

done.
Traceback (most recent call last):
  File "spawn_npc.py", line 104, in main
    world = client.get_world()
RuntimeError: time-out of 10000ms while waiting for the simulator, make sure the simulator is ready and connected to 127.0.0.1:2000

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "spawn_npc.py", line 291, in <module>
    main()
  File "spawn_npc.py", line 277, in main
    client.apply_batch([carla.command.DestroyActor(x) for x in vehicles_list])
RuntimeError: rpc::timeout: Timeout of 10000ms while connecting to 127.0.0.1:2000

the output of my sudo ufw status: 
Status: active

To                         Action      From
--                         ------      ----
2000/tcp                   ALLOW       Anywhere
2001/tcp                   ALLOW       Anywhere
2000/tcp (v6)              ALLOW       Anywhere (v6)
2001/tcp (v6)              ALLOW       Anywhere (v6)

can someone help me out about this? why the timeout issue is happened?  thanks in advance.
PS: i just update my carla to the newest version(github master version)

GabiShahmayster, 2020-08-03:
Hi @gengtsh,

Sorry about the silly question but is the CARLA server actually running?
Because your errors indicate that the client is unable to reach the server (which should be running on the same machine, using port 2000, as your client's attempted connection to 127.0.0.1:2000 indicates)

Gabi

germanros1987, 2020-08-19:
@gengtsh is your issue still bothering you?

closed 2020-11-08 by stale[bot]