carla-simulator/carla #3156: Carla docker image latest not starting the simulation

dHonerkamp, 2020-08-03:
Hi, sorry to bug you about this.

I've been trying to run the latest docker image, but it seems to be missing something. While I could run the previous versions (namely 0.9.5 and 0.9.9) successfully, the latest one simply returns to the console after the following printout:
```
4.24.3-0+++UE4+Release-4.24 518 0
Disabling core dumps.
sh: 1: xdg-user-dir: not found
```

(The same warning appears in the older versions,  but those then proceed to start the simulation).

Command to run the image:
```
docker pull carlasim/carla:latest
docker run --gpus all -it carlasim/carla:latest
```
same with passing the command (with and without -vulkan flag) and running it inside the container.
```
docker pull carlasim/carla:latest
docker run --gpus all carlasim/carla:latest /bin/bash CarlaUE4.sh -vulkan
```

Any help would be much appreciated.

thytless, 2020-08-04:
Hello! Try `-opengl` instead of `-vulkan`?
I also experienced same issue and the above works for me.
However I am also wondering why `-vulkan` not works, since we are using docker.

dHonerkamp, 2020-08-04:
Thanks for your reply. Just tried that, but same result: the command seems to return (and the container exits) after a few seconds without actually starting the simulation. Did you use the current :latest image when it worked for you?

fabianoboril, 2020-08-04:
@dHonerkamp: Same issue here. Let me pull in some experts.

thytless, 2020-08-04:
@dHonerkamp I checked your command again and I found that there is an option missing '--runtime=nvidia'. See https://carla.readthedocs.io/en/latest/build_docker/
Is it intended?

dHonerkamp, 2020-08-04:
@thytless Docker integrated gpu support, so the runtime argument isn't needed anymore. 
Are docker image IDs comparable? If so, you are not using the one I'm refering to. I have `6923977bc7dc` (digest `08780bee1d6b940c15479ecd135327ed34cdf4bd15ecb8b2a35f70283ec63636`)

thytless, 2020-08-04:
@dHonerkamp I guess the first one is digest, and sorry for not having made it clear. The second `7b45ce9` is git reversion of master branch, I hope this could help you find some diff!

germanros1987, 2020-08-05:
hi @dHonerkamp,

latest is now pointing to 0.9.10-pre, which is having issues with the OpenGL backend. You can only run it using the vulkan backend. However, to run CARLA with the vulkan backend you need to make some changes.

r-snijders, 2020-08-05:
I had a similar issue just this night. The following works for me (on Ubuntu 20.04LTS, NVidia driver 440.100, GTX1080):

```
xhost +
docker run -e SDL_VIDEODRIVER=x11 -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -p 2000-2002:2000-2002 -it --gpus all carlasim/carla:0.9.9 ./CarlaUE4.sh -opengl
```

dHonerkamp, 2020-08-06:
@germanros1987 thanks, that helped! Passing the Xorg socket did the trick :)

closed 2020-08-06 by dHonerkamp

kimown, 2020-08-12:
> hi @dHonerkamp,
> 
> latest is now pointing to 0.9.10-pre, which is having issues with the OpenGL backend. You can only run it using the vulkan backend. However, to run CARLA with the vulkan backend you need to make some changes.
> 
> First, you need to make sure that you invoke carla with the -vulkan flag. You also need to use the [latest dockerfile](https://github.com/carla-simulator/carla/blob/master/Util/Docker/Release.Dockerfile). The one that includes the libvulkan1 dependency and these modifications:
> 
> ENV SDL_VIDEODRIVER=x11
> CMD /bin/bash CarlaUE4.sh -vulkan
> 
> In addition, the way you launch the container is different for vulkan. You need to pass the Xorg socket:
> 
> `sudo -E docker run --privileged --rm --gpus all -it --net=host -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:rw carla:latest /bin/bash`
> 
> Hope this helps.

`DISPLAY=$DISPLAY` My server doesn't have a physical display, can I use X.org generate virtual display in host machine and use it?

fabiankueppers, 2021-01-05:
As @kimown mentioned, my server runs headless as well. Do you know a solution for headless mode using the docker files for versions 0.9.10 and above?

pavanteja295, 2021-01-12:
> hi @dHonerkamp,
>
> latest is now pointing to 0.9.10-pre, which is having issues with the OpenGL backend. You can only run it using the vulkan backend. However, to run CARLA with the vulkan backend you [...]

Hello @germanros1987.  The above solution still throws the same error as @dHonerkamp  pointed out with carla0.9.11 in docker and no physical display. Could you please help me in getting it working?