carla-simulator/carla #3160: change vehicle size in carla-0.9.9

whuxhzhang, 2020-08-04:
Carla-0.9.9 current vehicle size is too small，the size deviation is larger than that of some real vehicles.
So I would like to ask if I can change the size of the carla-0.9.9 vehicle.

Thanks.

germanros1987, 2020-08-19:
@whuxhzhang what do you mean the vehicle size is too small? All vehicles we have created have real dimensions and properties. Which vehicle is too small?

whuxhzhang, 2020-08-20:
> @whuxhzhang what do you mean the vehicle size is too small? All vehicles we have created have real dimensions and properties. Which vehicle is too small?

Ok, thanks for your replay.
I need a larger size model of the car，the body is 6m long and 2.24m wide, but not yet in CARLA, so I want to change the vehicle size.

closed 2020-11-08 by stale[bot]