carla-simulator/carla #3161: Besides using RoadRunner, is there any free software that allows me to generate carla maps？

XieJinLiang, 2020-08-04:
Besides using RoadRunner, is there any free software that allows me to generate carla maps? I want to import my defined map in carla, but RoadRunner has to charge. I know that now have carla0.9.10 version, but it seems still can not get, or I can be directly converted into a map with osm map of opendrive.

germanros1987, 2020-08-19:
Sorry for the delay, we are in the middle of vacation season. 

@XieJinLiang that is a good question. Let me describe the current state of maps in CARLA.

CARLA uses OpenDRIVE + FBXs to represent the maps. If you find a tool that can produce OpenDRIVE and you have the FBX associated to the geometry of the scene then you are ready to go. We provide the tools to import those objects into CARLA.
Now... we know that there are not many open source tools to generate OpenDRIVE maps. We are planning to work on this eventually (unless somebody decides to make this tool first).

In the meantime, what can you do? Well, you could try to use the new OSM generation tool that we added in 0.9.10-pre.  It's not a full solution yet, but it can generate the road layout for traffic simulation purposes.

You can get a trial license for RoadRunner too, or if you are a researcher at a public institution you can ask for a research license.

Sorry for the invonveniences.

revikiran1991, 2020-08-19:
@germanros1987  Can you please give me the link for branch 0.9.10-pre? I am not able to find it.

revikiran1991, 2020-08-19:
@XieJinLiang How did you get carla 0.9.10?
@germanros1987 How to get carla 0.9.10-pre? Can guys please share a link?

XieJinLiang, 2020-08-19:
> @XieJinLiang How did you get carla 0.9.10?
> @germanros1987 How to get carla 0.9.10-pre? Can guys please share a link?

It seems not available yet. You can only see the new features of version 0.9.10 in the [CHANGELOG.md](https://github.com/carla-simulator/carla/blob/master/CHANGELOG.md#carla-099)

XieJinLiang, 2020-08-24:
> @XieJinLiang How did you get carla 0.9.10?
> @germanros1987 How to get carla 0.9.10-pre? Can guys please share a link?

[0.9.10-pre](https://github.com/carla-simulator/carla/blob/master/Docs/download.md)

b-schwab, 2020-09-01:
This doesn't exactly solve your problem, but could potentially be interesting:

With [rtron](https://github.com/tum-gis/rtron) you can convert your OpenDRIVE datasets to CityGML (semantic city model standard).
The CityGML models can then be processed and modified with the software [FME](https://www.safe.com/fme/) by dragging and dropping a whole variety of transformers. FME also provides a writer towards the [Unreal Engine](https://www.safe.com/webinars/your-data-in-unreal-how-to-bring-your-data-into-real-time-environments/).
With the workflow, it is relatively easy to e.g. swap OpenDRIVE road objects with high-resolution models, or simply bring only the colored lane boundaries into the visualization.

I've listed some demos on [this page](https://rtron.io):
[image]

XieJinLiang, 2020-09-01:
@b-schwab Thank you very much, 
I think this is very interesting and it seems very helpful for my research work

closed 2020-11-14 by stale[bot]