carla-simulator/carla #3162: Impossible to import map from Roadrunner containing traffic lights or any other props

salvatoresan, 2020-08-04:
Hi, I'm having some issues importing map generated from Roadrunner. I'm working on the source version of carla 0.9.9 on ubuntu and i 've tried both the make import process and the Roadrunner plugin process with the last one able to import all the textures. Both processes work fine when the map to be imported contains only roads and grass. When i add even one traffic light or tree in the map, the import process fails causing UE4 crash. Any idea about this issue?

germanros1987, 2020-08-19:
Sorry for the delay, we are in the middle of the vacation season. @marcgpuig could you have a look into this? Perhaps is just a matter to use 0.9.9.4 or 0.9.10-pre...

rowandempster, 2020-09-17:
@salvatoresan I have the same issue, the make import command prints out lots of errors about not being able to find traffic light related assets.

rowandempster, 2020-09-18:
To be more specific, using the "deprecated" method with the roadrunner plugins works (traffic lights show up). But using the `make import` on build from source **and** has errors about missing assets and doesn't render traffic lights.

closed 2020-12-13 by stale[bot]

CoderTylor, 2021-03-18:
> Hi, I'm having some issues importing map generated from Roadrunner. I'm working on the source version of carla 0.9.9 on ubuntu and i 've tried both the make import process and the Roadrunner plugin [...]

hi there I have also met this problem ,can you contact me wechat:939490967 or email tylorwang@bupt.edu.cn

dilloncsr, 2023-03-16:
> > Hi, I'm having some issues importing map generated from Roadrunner. I'm working on the source version of carla 0.9.9 on ubuntu and i 've tried both the make import process and the Roadrunner plugin process with the last one able to import all the textures. Both processes work fine when the map to be imported contains only roads and grass. When i add even one traffic light or tree in the map, the import process fails causing UE4 crash. Any idea about this issue?
> 
> hi there I have also met this problem ,can you contact me wechat:939490967 or email [tylorwang@bupt.edu.cn](mailto:tylorwang@bupt.edu.cn)

Were you ever able to figure this out? I know this is an old thread but I am currently using 0.9.14 and still having the same issues as others... Would love to get this figured out ASAP.

Hanss8, 2023-04-22:
I have this problem too. When i import my map with Roadrunner plugins, the traffic lights occurs, there is problem that they don't work. Please can someone help us with this problem? Thanks.

dilloncsr, 2023-04-22:
> I have this problem too. When i import my map with Roadrunner plugins, the traffic lights occurs, there is problem that they don't work. Please can someone help us with this problem? Thanks.

I think this is an easy fix for what you already experiencing. When you create the map in RR, make sure that you signalize the intersection. You can do auto signalize and auto detect signals. This creates a controller from RR that Carla will use to control the lights.

dilloncsr, 2023-04-24:
> Thanks @dilloncsr <https://github.com/dilloncsr>. This helped. But when I
> started simulation in Unreal, the red light is still on and not switching
> to orange and green. Can you help me with this please?

In RR set the signalization to Split Phase and export the xodr again. You
can set custom timing if you like but the standard setting with split phase
works.

lemon-lone, 2024-08-08:
Has anyone encountered the problem of traffic signs not displaying after I imported the map

SExpert12, 2024-08-16:
Yes. I did have that and then I used the carla doc and add the traffic lights said in carla doc. Traffic light are included in actor list but it is not working. I have also raised an issue in carla for the same. 

Thanks