carla-simulator/carla #3163: make PythonAPI failed: recipe for target 'osm2odr' failed

sissiok, 2020-08-04:
When I run "make PythonAPI", I failed and got the following error message:

```
Setup.sh: llvm-8.0 already installed.
Setup.sh: boost-1.72.0-c8 already installed.
Setup.sh: rpclib-v2.2.1_c3-c8 already installed.
Setup.sh: gtest-1.8.1-c8 already installed.
Setup.sh: recast-cdce4e-c8 already installed.
Setup.sh: CARLA version 0.9.9.4-255-gc56a7738.
Version.h
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Client.Release" configuration.
[5/5] Linking CXX executable LibCarla/cmake/test/libcarla_test_client_release
[0/1] Install the project...
-- Install configuration: "Client"
-- Installing: /home/xiruo/carla/PythonAPI/carla/dependencies/include/carla/Version.h
-- Installing: /home/xiruo/carla/PythonAPI/carla/dependencies/lib/libcarla_client.a
-- Installing: /home/xiruo/carla/PythonAPI/carla/dependencies/test/libcarla_test_client_release
-- Set runtime path of "/home/xiruo/carla/PythonAPI/carla/dependencies/test/libcarla_test_client_release" to ""
BuildLibCarla.sh: Success!
-- Eclipse version is set to 3.6 (Helios). Adjust CMAKE_ECLIPSE_VERSION if this is wrong.
-- CMAKE_BINARY_DIR: /home/xiruo/carla/Build/libosm2dr-build
-- CMAKE_SOURCE_DIR: /home/xiruo/carla/Util/OSM2ODR
-- 
-- Platform: 
--     Host: Linux-5.4.0-42-generic x86_64
--     Target: Linux-5.4.0-42-generic x86_64
--     CMake: 3.10.2
--     CMake generator: Ninja
--     CMake build tool: /usr/bin/ninja
--     Compiler: Clang 8.0.1
-- 
-- Found Python: /usr/bin/python
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Failed to find XercesC (missing: XercesC_LIBRARY XercesC_INCLUDE_DIR
  XercesC_VERSION)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.10/Modules/FindXercesC.cmake:91 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:88 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/xiruo/carla/Build/libosm2dr-build/CMakeFiles/CMakeOutput.log".
Util/BuildTools/Linux.mk:158: recipe for target 'osm2odr' failed
make: *** [osm2odr] Error 1
```

What is going on here? My system is as below

Carla version: 0.9.9
OS: Ubuntu 18.04
Python on the machine: 2.7 & 3.6

elvircrn, 2020-08-10:
`apt install libxerces-c-dev` could be a work-around.

charlie-wartnaby, 2020-08-17:
I have hit the same problem attempting to build the docker image (using Ubuntu 18.04 as my 'host' OS) using the instructions at https://github.com/carla-simulator/carla/tree/master/Util/Docker . I'm building from recent (0.9.9) source, commit 8557c0b4.

I'm trying your tip of adding libxerces-c-dev to the list of installs in Prerequisites.Dockerfile. I'll report back when it's done (takes a while!).

charlie-wartnaby, 2020-08-18:
OK I added  libxerces-c-dev to the list of apt-install packages in `Prerequisites.Dockerfile` which did indeed get me past the Xerces problem, thanks.
Next my docker build failed because the Python module `distro` was not found. I added `pip` and `pip3 install distro` to the `Carla.Dockerfile` command to get past that. (Would have been more elegant to add it to Prerequisites, but I didn't want to have to run all of that again.)
I don't understand why anyone else should have been able to build the Docker image without those changes -- I thought the whole point was that it gave a uniform environment for everyone? But it's my first experience with Docker.
I have not succeeded yet though. My Carla.Dockerfile build still failed with this:
```
  LogTexture: Display: Building textures: T_Jacket_n (BC5, 2048X2048)
  LogTexture: Display: Building textures: T_Jacket_orm (AutoDXT, 2048X2048)
Took 795.927096s to run UE4Editor, ExitCode=137
ERROR: Cook failed.
       (see /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt for full exception trace)
AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
RunUAT ERROR: AutomationTool was unable to run successfully.
make: *** [package] Error 25
Util/BuildTools/Linux.mk:16: recipe for target 'package' failed
```
I have plenty of spare disk space by the way, and 32GB of RAM, so I don't think it was running out of resources. I will try and figure out how to access the logfile in the docker image (if that got built despite the command error?) to figure out that next problem.

charlie-wartnaby, 2020-08-18:
I tried running that docker build with a specific memory limit of 24GB using -m, no change.
And emitted the exception trace after it failed:
```
InternalUtils.SafeDeleteFile: SafeDeleteFile /home/ue4/UnrealEngine/Engine/Programs/AutomationTool/Saved/Cook-2020.08.18-13.31.10.txt
Log.WriteException: ==============================================================================
Log.WriteException: ERROR: Cook failed.
Log.WriteException:        (see /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt for full exception trace)
Log.WriteException: 
Log.WriteException: CommandletException: Editor terminated with exit code 137 (signal 9) while running Cook for /home/ue4/carla/Unreal/CarlaUE4/CarlaUE4.uproject; see log /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Cook-2020.08.18-13.44.38.txt
Log.WriteException:   at AutomationTool.CommandUtils.RunCommandlet (Tools.DotNETCommon.FileReference ProjectName, System.String UE4Exe, System.String Commandlet, System.String Parameters, System.String& DestLogFile) [0x0076d] in <21363ae3906c428bb150fffa2029575d>:0 
Log.WriteException:   at AutomationTool.CommandUtils.RunCommandlet (Tools.DotNETCommon.FileReference ProjectName, System.String UE4Exe, System.String Commandlet, System.String Parameters) [0x00001] in <21363ae3906c428bb150fffa2029575d>:0 
Log.WriteException:   at AutomationTool.CommandUtils.CookCommandlet (Tools.DotNETCommon.FileReference ProjectName, System.String UE4Exe, System.String[] Maps, System.String[] Dirs, System.String InternationalizationPreset, System.String[] CulturesToCook, System.String TargetPlatform, System.String Parameters) [0x0013e] in <21363ae3906c428bb150fffa2029575d>:0 
Log.WriteException:   at Project.Cook (AutomationTool.ProjectParams Params) [0x007ba] in <7a3f86f981b2444c826d69d05b013dc4>:0 Wrapped by AutomationException: Cook failed.
```
I don't know if that's useful to anyone. Has anyone actually succeeded in building the docker image recently, and on Ubuntu 18.04? (The docs say it was tested under 16.04, though it uses 18.04, i.e. as the Docker OS.)

germanros1987, 2020-08-19:
Sorry for the delay, we are in the middle of the vacation season. @Axel1092 could you please have a look at this?

charlie-wartnaby, 2020-08-19:
Thank you @germanros1987 ! I don't really understand the nature of the (permanent?) CARLA team, but while I don't expect any formal support of course, I hugely appreciate anyone looking into issues which affect me.
In fact I've run the 'make package' process that would have been the end result of doing the Docker-based map ingestion, and I still don't end up with a map that sprouts traffic lights like those shown in Bernat's very helpful video at https://www.youtube.com/watch?v=mHiUUZ4xC9o . So while it would be great to fix the docker process for others, it is not my personal roadblock right now.
I also added to #3086 while pursuing that effort, but I now think that RecastBuilder doesn't help with the traffic lights either.
I guess I should track down an existing issue or raise a new one to address the fundamental problem for me -- that I have tried all the import methods available to me, and so far failed to get CARLA-addressable traffic lights from my own RoadRunner model. And it would help so much if there was at least one very simple model available in .fbx/.xodr form that is believed to work to try out (or even .rrscene, but licensing of associated assets is a problem I guess with sharing the RoadRunner original).
For context, I am working on a proving ground model for CAV use, where we will deploy connected traffic lights, and want to get to the point of running simulation scenarios (via scenario_runner) that uses those lights. But I have totally failed to get traffic lights working in my model, except by the deprecated RoadRunner plug-in which does not set them up as CARLA objects in the same way. We are keen on CARLA because as a proving-ground provider, we will want to support future clients who may have bought into other simulation toolchains (or none), and would rather start with a licence-free solution for them to build on, which CARLA seems a great answer for.
Thanks again!
Charlie Wartnaby, Applus IDIADA, UK.

charlie-wartnaby, 2020-08-24:
On the traffic light front by the way, in the end I gave up on trying to import them from my RoadRunner map. I never did see the run-time autocreation of traffic lights shown in https://www.youtube.com/watch?v=mHiUUZ4xC9o . The only thing that worked for me was deleting any RoadRunner-designed traffic lights, and creating new ones from CARLA blueprints, and adding those to a traffic light group (it crashed if I didn't add them to a group). So I got there in the end. Just noting this in case anyone else finds this issue googling about traffic light imports.

rowandempster, 2020-09-15:
Hi @charlie-wartnaby, I am currently in the exact same boat as you: I want to make a Carla package for a RoadRunner map that I have exported. However, the only importation method of the RoadRunner export that "works" (things render correctly) is the deprecated UE4 Plugin method, which does not generate standalone packages and therefore it is difficult/impossible to distribute the map I have created.

If you have made any more progress on packaging RoadRunner exports and getting them to render correctly in UE4 please let me know :)

charlie-wartnaby, 2020-09-15:
Hi @rowandempster , I stopped at the point I had working traffic lights (added in the UE Editor from the blueprint library). For my project, just sharing the top-level ?.umap file and folder of asset files is good enough, so I fear I didn't pursue the package-making process. (I'm afraid I'm on holiday so I don't have access to any of the files right now.)
I did get one slightly random OpenDrive-derived traffic light popping up (in a junction with 4 lights) after all, which at this point I didn't really want! It appears by magic using OpenDrive data somehow, but how it happens I have not tracked down yet, nor seen any documentation for.
But see also #3189 where I found the RecastBuilder tool quietly did nothing -- well maybe it wrote to a logfile I haven't yet identified, but in general it exits quietly with no evidence on the command line when things go wrong.

closed 2020-12-25 by stale[bot]

tungnt101294, 2021-06-27:
> OK I added libxerces-c-dev to the list of apt-install packages in `Prerequisites.Dockerfile` which did indeed get me past the Xerces problem, thanks.
> Next my docker build failed because the Python module `distro` was not found. I added `pip` and `pip3 install distro` to the `Carla.Dockerfile` command to get past that. (Would have been more elegant to add it to Prerequisites, but I didn't want to have to run all of that again.)
> I don't understand why anyone else should have been able to build the Docker image without those changes -- I thought the whole point was that it gave a uniform environment for everyone? But it's my first experience with Docker.
> I have not succeeded yet though. My Carla.Dockerfile build still failed with this:
> 
> ```
>   LogTexture: Display: Building textures: T_Jacket_n (BC5, 2048X2048)
>   LogTexture: Display: Building textures: T_Jacket_orm (AutoDXT, 2048X2048)
> Took 795.927096s to run UE4Editor, ExitCode=137
> ERROR: Cook failed.
>        (see /home/ue4/Library/Logs/Unreal Engine/LocalBuildLogs/Log.txt for full exception trace)
> AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
> RunUAT ERROR: AutomationTool was unable to run successfully.
> make: *** [package] Error 25
> Util/BuildTools/Linux.mk:16: recipe for target 'package' failed
> ```
> 
> I have plenty of spare disk space by the way, and 32GB of RAM, so I don't think it was running out of resources. I will try and figure out how to access the logfile in the docker image (if that got built despite the command error?) to figure out that next problem.
> Incidentally, it is extremely painful having the 12GB assets download as part of that last stage -- if something goes wrong, it means doing it all over again. If that download could be made a separate step to get the local file, and have that ingested into the docker build, it would be a great advantage.

@charlie-wartnaby  Hi. Have you been able to find a solution to this issue?

charlie-wartnaby, 2021-06-28:
> @charlie-wartnaby Hi. Have you been able to find a solution to this issue?

I'm sorry @tungnt101294 , I haven't looked at it since. This was back in CARLA 0.9.9 or 0.9.10.

tungnt101294, 2021-06-28:
> > @charlie-wartnaby Hi. Have you been able to find a solution to this issue?
> 
> I'm sorry @tungnt101294 , I haven't looked at it since. This was back in CARLA 0.9.9 or 0.9.10.

@charlie-wartnaby Thanks for the reply.

I'm building CARLA 0.9.10. If you dont mind my asking, what did you do? Building another version?

charlie-wartnaby, 2021-06-28:
Hi @tungnt101294 , I just gave up on using Docker. Instead I just run carla using 'make launch' (and then 'make launch-only').

I was trying out the Docker build as that was a recommended route for importing a new map, but in the end the only way I found I could usefully import my map from RoadRunner was to go through the deprecated instructions for importing the assets manually into Unreal Editor.

tungnt101294, 2021-06-29:
@charlie-wartnaby I was building CARLA 0.9.10 on two PCs. One (WSL2 Ubuntu 18.04, 8GB RAM) failed and produced the exact same error while the other (Manjaro 5.4.123.1, 16GB) was actually successful. So I suspected the former just didnt have enough memory.

charlie-wartnaby, 2021-06-29:
> @charlie-wartnaby I was building CARLA 0.9.10 on two PCs. One (WSL2 Ubuntu 18.04, 8GB RAM) failed and produced the exact same error while the other (Manjaro 5.4.123.1, 16GB) was actually successful. So I suspected the former just didnt have enough memory.

Ah interesting -- I mean my first errors were missing dependencies or something, which just looked wrong no matter what resources the PC had. But my eventual failure was on a 32 MB Ubuntu 18.04 machine. Maybe it's an Ubuntu thing rather than RAM.

muhuzhongxun, 2022-12-20:
所以还是没人回答 @sissiok  的问题