carla-simulator/carla #3164: How to build Carla 0.9.9 on Ubuntu 20.04?

DavidLzr, 2020-08-05:
Hello Guys, I am a beginner of Carla.
I am here to seek for your help about building Carla on my Ubuntu 20.04 system. The guide provided in the Official Documentation https://carla.readthedocs.io/en/latest/build_linux/ says the Build can only be done on 18.04 and previous versions. Is there a way to build Carla on 20.04? 
Thank you for any possible help : )

r-snijders, 2020-08-06:
Hello, I didn't trying building Carla, but instead of building Carla from scratch, you can also consider using Docker and depend on the official carlasim images. Which I prefer, it allows me to deploy Carla on different systems without messing with the hosts dependencies etc.

Here are some notes I made for myself, I hope it also helps you in case you want to use Docker. These instructions should work if you have a nvidia card and nvidia-docker (https://github.com/NVIDIA/nvidia-docker) installed. (I am still working on a Dockerfile that depends on carlasim/carla:latest).

For Carla 0.9.9 on Ubuntu 20.04LTS with a GTX1080 (and , the following works:
* `docker run -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -p 2000-2002:2000-2002 -it --gpus all carlasim/carla:0.9.9 bash`
* Inside container: `SDL_VIDEODRIVER=x11 ./CarlaUE4.sh -opengl`

Python API usage
* By default, Carla ships with and egg (`/opt/carla-simulator/PythonAPI/carla/dist/carla-0.9.9-py3.7-linux-x86_64.egg`) which requires Python3.7 (also Python2.7, but please don't use it, Python2 is deprecated!).
* The pip3 version is outdated (0.9.5).
* Use `add-apt-repository ppa:deadsnakes/ppa` in the container to install the right python version for arbitrary ubuntu versions (e.g., python3.7 on ubuntu 16.04), otherwise the egg won't work.
* Extra dependencies include: curl, pip3.7, libpng16-dev, libjpeg8-dev, libtiff5-dev
* `strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep CXXABI` reveales that the required `CXXABI_1.3.11` is not available by default in the docker images using ubuntu 16.04
  * Solution:
  * `add-apt-repository ppa:ubuntu-toolchain-r/test`
  * `apt-get upgrade libstdc++6`
  * Then `python3.7 config.py -x opendrive/TownBig.xodr` from within the container works.

germanros1987, 2020-08-19:
Thanks @r-snijders for taking care of this!

closed 2020-08-19 by germanros1987

ryanbranch, 2020-10-26:
Hello @r-snijders, thanks for this write-up!

Just curious if you've since created the aforementioned Dockerfile that depends on carlasim/carla:latest? If so, can I access it somewhere on your GitHub?