carla-simulator/carla #3166: Error when running "make PythonAPI"

sulichuan, 2020-08-05:
Hi~
I tried to build Carla in Linux, and followed the steps to build.
The process went smoothly until running "make PythonAPI" and the script shows error message 
the message is : 
ImportError: No module named distro
Util/BuildTools/Linux.mk:87: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1

I tried to install distro with "pip3 install distro", and it still shows the same message
the message after install pip3 is : 
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: distro in /usr/local/lib/python3.7/dist-packages (1.5.0)

Is anyone knows how to solve this situation?

RehabHsn, 2020-08-12:
The same error here, have you found a solution?

qhaas, 2020-08-18:
I haven't seen this particular error, but it might be a good idea to make sure you have the same dependencies/packages deployed in your development environment as [CARLA's build environment](https://github.com/carla-simulator/carla/blob/0.9.9.4/Util/Docker/Prerequisites.Dockerfile) (see the packages they `apt-get install`), especially `python3-dev` and `python-dev`.

germanros1987, 2020-08-19:
Sorry for the delay, we are in the middle of the vacation season. @marcgpuig could you help with this?

EsquilliDev, 2020-10-22:
Having the same issue on Ubuntu 20.04 carla:latest

philippwulff, 2020-10-22:
I am facing the same error when installing on Linux Mint 19.3. Has anyone solved it?

EsquilliDev, 2020-10-22:
@philippwulff For some reason, while doing the dependencies part, one command didn't run correctly. Just run this command to fix it

Command
`pip3 install --user distro`

closed 2021-01-26 by germanros1987